gdsfactory/gdsfactory #1124: How to change width of spiral_double component

ajadk, 2023-01-10:
When creating the spiral_double component, there is no argument to specify the waveguide width. 

Looking into the source, the component is created by extruding a spiral_archimedian path, but no width argument is passed to the extrude function. Same is true when creating the two bends in the middle. 

This can be fixed by simply adding a width argument that is passed to the bend constructor and the extrude function for the spiral.

joamatab, 2023-01-10:
Hi,

You can modify the cross-section
https://gdsfactory.github.io/gdsfactory/components.html#spiral-double

as explained here
https://gdsfactory.github.io/gdsfactory/notebooks/08_pdk.html#CrossSectionSpec

```
import gdsfactory as gf

c = gf.components.spiral_double(
    min_bend_radius=10.0,
    separation=2.0,
    number_of_loops=3,
    npoints=1000,
    cross_section=dict(cross_section="strip", settings=dict(width=1)),
)
c.show()
```

ajadk, 2023-01-10:
Thank you very much! I'm sorry I'm new to the cross section concept.

joamatab, 2023-01-10:
Yes 👍 it's a new thing, let me know if you find any further improvements

closed 2023-01-10 by ajadk