gdsfactory/gdsfactory #1134: create xsection with that is not constant?

joamatab, 2023-01-11:
How could we create a cross-section that is not constant, for example, that has vias every 10um?

something like this

[image]

simbilod, 2023-01-11:
I could use something like this too

I tried something quick, do you know why I could be getting the extra path on top?

[image]

Here's the new function:

`@cell
def extrude_segments(
    p: Path,
    cross_sections = None,
    segment_lengths = None,
    layer: Optional[LayerSpec] = None,
    simplify: Optional[float] = None,
    shear_angle_start: Optional[float] = None,
    shear_angle_end: Optional[float] = None,
) -> Component:
    """Returns Component extruding a Path with multiple cross_sections.

    Like extrude, but splits the path in segments of length segment_lengths, each associated with a cross_section in cross_sections.

    Args:
        (See extrude)
        cross_sections: list of CrossSectionSpec 
        segment_lengths: list of segment lengths

    """
    from shapely.ops import substring
    from shapely.geometry import LineString

    c = Component()

    path_linestr = LineString(p.points)
    cumul_dist = 0
    completed = False
    while not completed:
        for segment_length, cross_section in zip(segment_lengths, cross_sections):
            if not completed:
                if cumul_dist + segment_length > p.length():
                    end_dist = p.length() - cumul_dist
                    completed = True
                else:
                    end_dist = cumul_dist + segment_length
                subpath_str = substring(path_linestr, start_dist=cumul_dist, end_dist=end_dist)
                subpath = Path(path=subpath_str.coords)
                c << extrude(subpath, cross_section=cross_section, simplify=simplify)
                cumul_dist += segment_length

    return c`

Here's the call:

`    import gdsfactory as gf
    points = np.array([(20, 10), (40, 10), (20, 40), (50, 40), (50, 20), (70, 20)])

    P = gf.path.smooth(
        points=points,
        radius=2,
        bend=gf.path.euler,  # Alternatively, use pp.arc
        use_eff=False,
    )

    # Add a few "sections" to the cross-section
    x = gf.CrossSection(
        width=0.5, offset=1, layer=(2, 0), port_names=("in", "out")
    )
    x2 = gf.CrossSection(
        width=0.5, offset=0.5, layer=(1, 0), port_names=("in", "out")
    )
    x3 = gf.CrossSection(
        width=0.5, offset=-0.5, layer=(1, 0), port_names=("in", "out")
    )

    c = gf.path.extrude_segments(P, cross_sections=[x,x2,x3], segment_lengths=[1.0, 0.5, 0.75])
    c.show(show_ports=True)`

95-martin-orion, 2023-01-27:
Thanks @simbilod for picking this up! Regarding the draft solution: I tried patching it into my local branch for use in a larger design, but found it to be rather slow. Would it be possible to cache and reuse the extruded geometry for straight segments somehow?

Rough metrics for my test:
- Three long wires with ~1500 segments each
- Three distinct `CrossSections` with at most two `Sections` each
- Total runtime: ~13 seconds

tvt173, 2023-02-20:
with the new all-angle router, you should be able to do this with a custom connector + bend
https://gdsfactory.github.io/gdsfactory/notebooks/042_non-manhattan-router.html#Bends-and-connectors

i.e. if you just make the connector extend the existing auto-taper connector + add an array of vias at the desired spacing, I think it should be much faster

95-martin-orion, 2023-10-23:
The solution to #1959 is adjacent to this, but not identical. It allows placing components along a path, but those components have no awareness of the path curvature.

joamatab, 2023-10-23:
agreed, i'll mark this issue as completed

https://gdsfactory.github.io/gdsfactory/notebooks/03_Path_CrossSection.html#option-2-arbitrary-cross-section

if there are any other cases we can create a new issue

closed 2023-10-23 by joamatab