gdsfactory/gdsfactory #1141: simplify cloud deployment

joamatab, 2023-01-13:
How could we simplify cloud deployment of gdsfactory?

For example, we could add https://github.com/proppy/rad-lab/tree/tuning/radlab-launcher

@proppy
@HelgeGehring 
@flaport

proppy, 2023-01-17:
I should already part of it, see: https://github.com/proppy/rad-lab/blob/tuning/modules/silicon_design/scripts/build/images/provision/environment.yml#L33

joamatab, 2023-01-17:
nice!

How about including a radlab documentation entry on the gdsfactory docs?

proppy, 2023-01-26:
@joamatab following up on the "one click"'ish deployment idea, I think it would possible to leverage https://github.com/marketplace/google-cloud-build with the https://github.com/proppy/rad-lab-deploy repo; rather than the rad-lab UI/CLI.

Do you want to give it a try?

closed 2023-11-02 by github-actions[bot]