gdsfactory/gdsfactory #1143: Time Domain Simulations - How to describe memory containing components

Jan-David-Black, 2023-01-14:
**Is your feature request related to a problem? Please describe.**
Similar to https://github.com/flaport/sax/issues/4 I would like to incorporate some way of performing time-domain simulations from gdsfactory netlists. However as far as I can see time domain simulations of passive networks can be relatively easily inferred from the S-parameters of the network: Either Fourier transform the input waveform, multiply by the respective frequency-dependent S-parameter and transform back (or do the whole story in time-domain by convolution). Am I mistaken?

Meaning that such simulations only really become challenging when involving memory containing elements (like in pythontorch/caphe etc.). How would we express these models most elegantly in the context of gdsfactory netlists / S-parameter models?

joamatab, 2023-01-14:
https://github.com/BYUCamachoLab/simphony/issues/56
https://github.com/TeledyneLeCroy/SignalIntegrity

@SkandanC 
@sequoiap

joamatab, 2023-01-19:
Time domain models:

- photodetector
- heater RC

joamatab, 2023-10-24:
Feel free to reopen in gplugins

https://gdsfactory.github.io/gplugins/notebooks/vlsir_netlist.html

closed 2023-10-24 by joamatab