gdsfactory/gdsfactory #1152: How can we reduce size of femwell modes?

joamatab, 2023-01-15:
Femwell modes are quite big for caching

How can we reduce size of femwell modes?

each simulation of 4 modes takes around 10MB of disk space

[image]

@HelgeGehring 
@simbilod

sebastian-goeldi, 2023-01-15:
If you can access and pass them as file descriptors, I'd suggest to store them in a tar.gz or tar.bz2 and access them through [tarfile](https://docs.python.org/3.10/library/tarfile.html#tarfile.TarFile.extractfile)

And example can be seen here https://stackoverflow.com/questions/37474767/read-tar-gz-file-in-python#37474813

simbilod, 2023-01-15:
The biggest thing would be to save the mesh instead of the basis function objects, and rebuild the basis when loading from the mesh + function arguments (with element type and order)

Obtenir Outlook pour Android<https://aka.ms/AAb9ysg>

joamatab, 2023-01-17:
Thank you for the PR Simon,

it still fails, you can run `simulation/fem/mode_solver.py` twice, 

the first time it will cache on your computer, the second one will fail

PR #1163 (2023-01-17): Fix femwell cache
fixes #1152
closed 2023-01-17 by joamatab