gdsfactory/gdsfactory #1157: support SAX simulations from schematic

joamatab, 2023-01-16:
How could we support SAX simulations from schematic?

@flaport 

```
---
jupyter:
  jupytext:
    text_representation:
      extension: .md
      format_name: markdown
      format_version: '1.3'
      jupytext_version: 1.14.4
  kernelspec:
    display_name: Python 3 (ipykernel)
    language: python
    name: python3
---

# Schematic Driven layout

The Schematic driven layout uses a schematic format similar to our `*.pic.yml`.

The Jupyter notebook interface allows you to get the best of both worlds of GUI and python driven based flows.

```python
from bokeh.io import output_notebook

import gdsfactory as gf
from gdsfactory.schematic_editor import SchematicEditor
from gdsfactory.config import rich_output

%env BOKEH_ALLOW_WS_ORIGIN=localhost:8888

output_notebook()
rich_output()
```

First you initialize a session of the schematic editor.
The editor is synced to a file.
If file exist, it loads the schematic for editing. If it does not exist, it creates it.
The schematic file is continuously auto-saved as you edit the schematic in your notebook, so you can track changes with GIT.

```python
se = SchematicEditor("test.schem.yml")
```

## Define instances

First you need to define which instances to include. We do this through this grid-like editor.
Components are auto-populated from your active PDK.

instance name | Component type
--------------| --------------
mmi1          | mmi1x2

```python
se.instance_widget
```

```python
se.instances.keys()
```

You can also **add your instances through code**, and since it is just a dictionary update, *the integrity of your schematic will be maintained, even after re-running the notebook* as-is.
You can here specify a component either by name or as an actual component, using auto-completion to specify your settings as well.

```python
se.add_instance("s1", gf.components.straight(length=20))
se.add_instance("s2", gf.components.straight(length=40))
```

But you can even query the parameters of default components, set only by name through the widget grid, like so:

```python
se.instances["mmi1"].settings.full
```

It is also possible to *instantiate through the widget, then set the settings of our component later, through code.*

By doing this through code, we have the full power of python at our disposal to easily use shared variables between components, or set complex Class or dictionary-based settings, without fumbling through a UI.

```python
se.update_settings("mmi1", gap_mmi=1.0)
se.update_settings("mmi2", gap_mmi=0.7)

for inst_name, inst in se.instances.items():
    if inst.settings.changed:
        print(f"{inst_name}: {inst.settings.changed}")
```

## Define nets

Now, you define your logical connections between instances in your netlist. Each row in the grid represents one logical connection.

```python
se.net_widget
```

Similarly, you can programmatically add nets.
Adding a net which already exists will have no effect, such that the notebook can be rerun without consequence.
However, trying to connect to a port which is already otherwise connected will throw an error.

```python
se.add_net(
    inst1="mmi1", port1="o2", inst2="s1", port2="o1"
)  # can be re-run without consequence
se.add_net(inst1="s1", port1="o1", inst2="mmi1", port2="o2")  # also ok
# se.add_net(inst1="s1", port1="o2", inst2="mmi1", port2="o2")  # throws error -- already connected to a different port
```

```python
se.schematic
```

## Visualize

You can visualize your schematic down below. After you've initialized the plot below, you should see it live-update after every change we make above.

Currently the *size of component symbols and location of ports are **layout-realistic***. This may be a nice default if you don't care to bother making symbols for your components.
But it would be a nice improvement for the future to allow associating symbols with components, to make the schematic easier to read.

If you activate the `Point Draw Tool` in the plot, you should see that you are able to arrange components freely on the schematic, and changes are saved back to the `*.schem.yml` file in realtime.

```python
se.visualize()
```

## generate Layout

You can use your schematic to generate a preliminary layout, and view in the notebook and/or KLayout. Initial placements come from schematic placements and Routes are auto-generated from nets.

```python
layout_filename = "sdl_demo.pic.yml"
se.instantiate_layout(layout_filename, default_router="get_bundle")
c = gf.read.from_yaml(layout_filename)
f = c.plot()
```

```python
# you can save your schematic to a standalone html file once you are satisfied
# se.save_schematic_html('demo_schem.html')
```

```python
netlist = c.get_netlist()

import numpy as np
import sax
import matplotlib.pyplot as plt

def straight(wl=1.5, length=10.0, neff=2.4) -> sax.SDict:
    wl0 = 1.5  # center wavelength for which the waveguide model is defined
    return sax.reciprocal({("o1", "o2"): jnp.exp(2j * jnp.pi * neff * length / wl)})

def mmi1x2():
    """Assumes a perfect 1x2 splitter"""
    return sax.reciprocal(
        {
            ("o1", "o2"): 0.5**0.5,
            ("o1", "o3"): 0.5**0.5,
        }
    )

def bend_euler(wl=1.5, length=20.0):
    """ "Let's assume a reduced transmission for the euler bend compared to a straight"""
    return {k: 0.99 * v for k, v in straight(wl=wl, length=length).items()}

models = {
    "bend_euler": bend_euler,
    "mmi1x2": mmi1x2,
    "straight": straight,
}

circuit, _ = sax.circuit(netlist=netlist, models=models)
```

```python
wl = np.linspace(1.5, 1.6)
S = circuit(wl=wl)

plt.figure(figsize=(14, 4))
plt.title("MZI")
plt.plot(1e3 * wl, jnp.abs(S["o1", "o2"]) ** 2)
plt.xlabel("λ [nm]")
plt.ylabel("T")
plt.grid(True)
plt.show()
```

```python

```
```

tvt173, 2023-01-16:
if i'm not mistaken, SAX always expects the format of an extracted netlist (all connections, no routes). so, as a first order solution, we could just interpret (or convert) all links in the schematic netlist as connections, then read with SAX as normal

tvt173, 2023-01-16:
that would give us an "idealized" circuit simulation, with no loss in the links. in the future, it may be nice to be able to stub the links with some assumed loss/s-matrix (i.e. assuming a waveguide of some length)

joamatab, 2023-01-16:
Hi Troy,

something is going on, when i run the notebook above i get a strange error

[image]

simbilod, 2023-01-16:
> Hi Troy,
> 
> something is going on, when i run the notebook above i get a strange error
> 
> [image]

Have you tried using `get_netlist_recursive` to break down the netlist into the models you gave?

joamatab, 2023-01-16:
yes, i tried that

it's missing ports

[image]

Troy, how can we add ports in the schematic view?

tvt173, 2023-01-16:
ah, i see. for schematics, you should need to define it, right? i may not have implemented that yet graphically through the notebook, but it should just be the same format as for pic.yml (`top_level_port: lower_level_inst:port`)

closed 2023-01-17 by joamatab