gdsfactory/gdsfactory #1164: CI/CD for plugins

simbilod, 2023-01-17:
How can we ensure CI/CD runs tests for the plugins?

Could we do something like this where if runs tests for specific plugins if the files have been changed?

https://www.codewrecks.com/post/azdo/pipeline/execution-condition-file-changed/

simbilod, 2023-01-17:
@joamatab

joamatab, 2023-01-17:
Hi Simon, 

I have fixed the issue of the plugins failing PRs,

You can see the code here
https://github.com/gdsfactory/gdsfactory/blob/fe5239807b26d76a09fe9d7fc3e3f41fe58831d6/.github/workflows/test_code.yml#L97

Now all the PRs should pass even if they are not made from gdsfactory branches

simbilod, 2023-01-17:
Thanks, hopefully this catches stuff like the femwell issue going forward :)

closed 2023-01-17 by simbilod