gdsfactory/gdsfactory #1170: speed up dev environment

joamatab, 2023-01-18:
How could we speed up gdsfactory dev environment?

https://github.dev/gdsfactory/gdsfactory

@basnijholt
@HelgeGehring 
@proppy

proppy, 2023-01-18:
I think having some colab environment for immediate testing could be good, and some conda based installer (like discussed with https://github.com/hdl/conda-eda/issues/245) for workstation installation.

wdyt?

joamatab, 2023-01-18:
I was playing with this and looks great

https://github.com/codespaces

proppy, 2023-01-18:
See also https://cloud.google.com/workstations, potentially related to #1141 #250

simbilod, 2023-01-19:
Note that the plugins do not come installed when launching a codespace -- is this intended?

[image]
[image]
[image]

Jan-David-Black, 2023-01-20:
As far as I can see codespace sharing is only possible within organizations. Are you able to access the codespace I created for https://github.com/gdsfactory/gdsfactory/issues/1181 : https://jan-david-black-probable-succotash-pq54456x6qx27v64.github.dev/

joamatab, 2023-01-20:
no, i can't 

for live sharing you can use VSCode extension

I have done pair programming using that

[image]

closed 2023-10-24 by joamatab