gdsfactory/gdsfactory #1178: Define testing directives, vectors, protocols

joamatab, 2023-01-19:


Jan-David-Black, 2023-01-19:
Idea of how to implement it:
A decorator used with the component, that contains the fully measurable circuit (`fmc`). The decorator identifies the grating couplers and electrical pads and labels them accordingly (including a common reference to the `fmc`). Additionally, it provides measurement ranges (e.g. current/voltage ranges incl. compliance and Fixed Wavelength/Wavelength ranges, optical power etc.) and routines (e.g. one slow sweep one fast sweep) in a directive. Going even further we might add directives to deembed the device characteristics.
The directives are added as annotations to the gds.

joamatab, 2023-01-20:
How about using the info settings from components?

```
    import gdsfactory as gf

    dft = dict(wavelength=1.55, polarization="te", hastags=["optical", "te"])
    c = gf.components.straight(info=dict(dft=dft))
    print(c.info)
    c.write_gds_with_metadata("a.gds")
    c.show(show_ports=True)
```

joamatab, 2023-10-24:
Documented here

https://gdsfactory.github.io/gdsfactory/notebooks/07_mask.html#metadata

closed 2023-10-24 by joamatab