gdsfactory/gdsfactory #1181: Error when trying to use the Grating coupler FDTD simulations notebook

SebyLeDanger, 2023-01-20:
Dear Sir/Madam,

### Error Description

I was trying to use the Grating coupler FDTD simulations notebook to train myself.

To do so, I am using VSCode with a virtual environment under Python 3.10.9. Another thing to note is that I have installed SiEPIC-Tools and SiEPIC-EBeam-PDK in Klayout.
I used pip with the following commands to setup the environment:

- pip install gdsfactory --upgrade
- pip install ubcpdk --upgrade
- gf tool install
- pip install tidy3d==1.8.2

I then tried to run the first cell of the notebook: 

`import numpy as np
import matplotlib.pyplot as plt

import gdsfactory.simulation as sim
import gdsfactory as gf
import gdsfactory.simulation.gtidy3d as gt

import ubcpdk
import ubcpdk.components as pdk

c = pdk.ebeam_gc_te1550()
c`

### Error output

And I got this error:

OSError                                   Traceback (most recent call last)
Cell In[4], line 11
      8 import ubcpdk
      9 import ubcpdk.components as pdk
---> 11 c = pdk.ebeam_gc_te1550()
     12 c

File d:\(2023-01,04) - University of Bristish Columbia _ Phot1x\Chapter n°01 - Passive Photonic Components\04 - Fibre Grating Coupler\.venv\lib\site-packages\gdsfactory\cell.py:161, in cell_without_validator.<locals>._cell(*args, **kwargs)
    156 if not callable(func):
    157     raise ValueError(
    158         f"{func!r} is not callable! @cell decorator is only for functions"
    159     )
--> 161 component = func(*args, **kwargs)
    163 # if the component is already in the cache, but under a different alias,
    164 # make sure we use a copy, so we don't run into mutability errors
    165 if id(component) in [id(v) for v in CACHE.values()]:

File d:\(2023-01,04) - University of Bristish Columbia _ Phot1x\Chapter n°01 - Passive Photonic Components\04 - Fibre Grating Coupler\.venv\lib\site-packages\pydantic\decorator.py:40, in pydantic.decorator.validate_arguments.validate.wrapper_function()

File d:\(2023-01,04) - University of Bristish Columbia _ Phot1x\Chapter n°01 - Passive Photonic Components\04 - Fibre Grating Coupler\.venv\lib\site-packages\pydantic\decorator.py:134, in pydantic.decorator.ValidatedFunction.call()

File d:\(2023-01,04) - University of Bristish Columbia _ Phot1x\Chapter n°01 - Passive Photonic Components\04 - Fibre Grating Coupler\.venv\lib\site-packages\pydantic\decorator.py:206, in pydantic.decorator.ValidatedFunction.execute()

File d:\(2023-01,04) - University of Bristish Columbia _ Phot1x\Chapter n°01 - Passive Photonic Components\04 - Fibre Grating Coupler\.venv\lib\site-packages\ubcpdk\components.py:401, in ebeam_gc_te1550()
...
---> 44     gdsii_lib = gdstk.read_gds(str(gdspath))
     45 elif gdspath.suffix.lower() == ".oas":
     46     gdsii_lib = gdstk.read_oas(str(gdspath))

OSError: Error opening input file.

### What I think I understand about the problem

It seems that I can't access the pdk components because of a bad gdspath, the question is why is this path wrong and where should the path leads in a correct installation ?

PS : I am new to gitHub and gdsfactory so I hope I am reporting my problem in the correct way and location.

Jan-David-Black, 2023-01-20:
Hey, I opened a codespace for this error with the intent to reproduce it: https://jan-david-black-probable-succotash-pq54456x6qx27v64.github.dev/ (I hope you can access it...)
I was however not able to reproduce the error. And the given code executes fine.
I have noticed you are using some deeply nested paths including some non-standard characters:
> University of Bristish Columbia _ Phot1x\Chapter n°01 - Passive Photonic Components\04 - Fibre Grating Coupler.venv\
Would you mind trying to install the environment in a location like your home folder or something like `~/photonics`?

Jan-David-Black, 2023-01-20:
Good to hear 👍 Would you mind closing the issue?

closed 2023-01-20 by SebyLeDanger