gdsfactory/gdsfactory #1203: Parasitics extraction from gds

simbilod, 2023-01-30:
For instance https://github.com/purdue-onchip/gds2Para

joamatab, 2023-01-31:
@proppy
@njcoburn

joamatab, 2023-01-31:
@das-dias
@proppy

proppy, you had a way to do this in magic

das-dias, 2023-04-23:
There is a way of doing this with magic and ngspice for spice netlist generation, but it requires an LVS and DRC clean layout design, otherwise the results magic gives might not be reliable - or there might not be any results at all.

Setup environment variables:
```shell
export PDK_PATH=path/to/pdk/sky130A

export GDS_FILE=path/to/gds/file.gds
export CELL_NAME=top_cell_name
export PEX_FILE=path/to/output/pex/file.spice
```

run_ext_pex.tcl : 
```tcl
# read design
gds read $::env(GDS_FILE)
load $::env(CELL_NAME)

# run netlist extraction for parasitic extraction (PEX) using magic
# extract for PEX
select top cell
port makeall
ext2spice lvs
ext2spice cthresh 0.01
ext2spice rthresh 0.01
ext2spice subcircuit on
ext2spice ngspice
ext2spice -o $::env(PEX_FILE)
```

Run the TCL script using magic:
```shell
magic -noconsole -dnull run_ext_pex.tcl -rcfile $PDK_PATH/libs.tech/magic/sky130A.magicrc
```

Append the skywater 130 spice library to the top of the generated spice file (in this case I am appending the "tt" library to perform simulations considering the typical corner of N and P devices ):
```shell
echo ".lib \"$PDK_PATH/libraries/sky130_fd_pr/latest/models/sky130.lib.spice\" tt" >> PEX_FILE
```

joamatab, 2023-10-24:
We should move this to gplugins

https://gdsfactory.github.io/gplugins/

closed 2023-10-24 by joamatab