gdsfactory/gdsfactory #1205: Need a page which just lists the plugins?

mithro, 2023-01-30:
I couldn't find a page which just lists all the plugins for GDSFactory. Currently it is hard to show off that you have the following plugins in one page. Something like;
 * [Plugins: Optimization](https://gdsfactory.github.io/gdsfactory/plugins_optimization.html)
     - [Ray Tune generic black-box optimiser](https://gdsfactory.github.io/gdsfactory/notebooks/plugins/ray/optimiser.html)
 * [Plugins: Device simulators](https://gdsfactory.github.io/gdsfactory/plugins_process.html)
     - [Thermal](https://gdsfactory.github.io/gdsfactory/notebooks/plugins/thermal/thermal.html)
     - [DEVSIM TCAD simulator](https://gdsfactory.github.io/gdsfactory/notebooks/plugins/devsim/01_pin_waveguide.html)
     - [Meshing](https://gdsfactory.github.io/gdsfactory/notebooks/plugins/devsim/01_pin_waveguide.html#Meshing)
     - xxxx
 * etc....

mithro, 2023-01-30:
FYI - @joamatab

SkandanC, 2023-01-30:
Hello Tim,

Here you can see the list of all the plugins https://gdsfactory.github.io/gdsfactory/

[image]

mithro, 2023-01-30:
I want a page which just list the plugins as that page has a lot of other things....