gdsfactory/gdsfactory #1210: A bug in gf.generic_tech.LAYER_VIEWS.preview_layerset()

HWOde, 2023-01-31:
I customized my PDK, defined Layer and Color, after ran the code below.

c=gf.generic_tech.LAYER_VIEWS.preview_layerset()

I got the picture below，we can seen that several layers are not displayed correctly, only blank can be seen.

[image]

At first, I thought there were some errors in my code, but after I adopted the default settings ( run the code below).

import gdsfactory as gf
c = gf.generic_tech.LAYER_VIEWS.preview_layerset()
c.plot(blocking=True,show_subports=False)

The same problem still occured, there are some layers missing in the picture.

[image]

At the same time, i notified that this problem also occurred in the document.

[image]

By the way, why does the background color of all the pictures in the document become black. This looks a little strange.

Version: 6.27.1
python: 3.9

sebastian-goeldi, 2023-01-31:
The document's color is black, because it now uses the LayoutView object from KLayout to shiw, and I believe it should be interactive (unless it broke). @tvt173 implemented it, unless I am wrong ;)

tvt173, 2023-01-31:
indeed, the background is black because it uses klayout now for the visualization, and with the headless klayout, the default is black. i prefer black, so it doesn't bother me, but i'm sure there is a way to change it if you would like to investigate and submit a PR!

indeed as sebastian says, the widget is still pretty primitive. it should support interactivity in the future, but today it probably does not work very well :)

sebastian-goeldi, 2023-01-31:
@tvt173 I believe the way to set the color would be by setting the `LayoutView().set_config("background-color", new_value)`. I am not sure what exactly it accepts, but probably worth a shot, though I also like black ;). All the configs can be acquired with `LayoutView.get_config_names()` which gives a list of the following settings:

```python
['absolute-units',
 'abstract-mode-enabled',
 'abstract-mode-width',
 'apply-text-trans',
 'background-color',
 'bitmap-caching',
 'bitmap-oversampling',
 'cell-list-sorting',
 'child-context-color',
 'child-context-dimming',
 'child-context-enabled',
 'child-context-hollow',
 'clear-ruler-new-cell',
 'color-palette',
 'combine-mode',
 'context-color',
 'context-dimming',
 'context-hollow',
 'current-lib-view',
 'current-ruler-template',
 'dbu-units',
 'default-add-other-layers',
 'default-font-size',
 'default-layer-properties',
 'default-text-size',
 'draw-array-border-instances',
 'drawing-workers',
 'drop-small-cells',
 'drop-small-cells-condition',
 'drop-small-cells-value',
 'edit-connect-angle-mode',
 'edit-grid',
 'edit-hier-copy-mode',
 'edit-inst-angle',
 'edit-inst-array',
 'edit-inst-cell-name',
 'edit-inst-column_x',
 'edit-inst-column_y',
 'edit-inst-columns',
 'edit-inst-mirror',
 'edit-inst-pcell-parameters',
 'edit-inst-place-origin',
 'edit-inst-row_x',
 'edit-inst-row_y',
 'edit-inst-rows',
 'edit-inst-scale',
 'edit-max-shapes-of-instances',
 'edit-move-angle-mode',
 'edit-path-ext-type',
 'edit-path-ext-var-begin',
 'edit-path-ext-var-end',
 'edit-path-width',
 'edit-pcell-show-parameter-names',
 'edit-show-shapes-of-instances',
 'edit-snap-to-objects',
 'edit-text-halign',
 'edit-text-size',
 'edit-text-string',
 'edit-text-valign',
 'edit-top-level-selection',
 'fit-new-cell',
 'flat-cell-list',
 'full-hierarchy-new-cell',
 'global-trans',
 'grid-axis-color',
 'grid-color',
 'grid-grid-color',
 'grid-ruler-color',
 'grid-show-ruler',
 'grid-style0',
 'grid-style1',
 'grid-style2',
 'grid-visible',
 'guiding-shape-color',
 'guiding-shape-line-width',
 'guiding-shape-vertex-size',
 'guiding-shape-visible',
 'hide-empty-layers',
 'highres-mode',
 'image-cache-size',
 'images-visible',
 'initial-hier-depth',
 'inst-color',
 'inst-label-font',
 'inst-label-transform',
 'inst-visible',
 'layers-always-show-layout-index',
 'layers-always-show-ld',
 'layers-always-show-source',
 'line-style-palette',
 'markers-visible',
 'min-inst-label-size',
 'mouse-wheel-mode',
 'no-stipple',
 'pan-distance',
 'paste-display-mode',
 'ruler-color',
 'ruler-grid-snap',
 'ruler-halo',
 'ruler-obj-snap',
 'ruler-snap-mode',
 'ruler-snap-range',
 'ruler-templates-v2',
 'rulers',
 'search-range',
 'search-range-box',
 'sel-color',
 'sel-dither-pattern',
 'sel-halo',
 'sel-inside-pcells-mode',
 'sel-line-width',
 'sel-transient-mode',
 'sel-vertex-size',
 'show-properties',
 'split-cell-list',
 'split-lib-views',
 'stipple-offset',
 'stipple-palette',
 'test-shapes-in-view',
 'text-color',
 'text-font',
 'text-lazy-rendering',
 'text-visible',
 'tracking-cursor-color',
 'tracking-cursor-enabled']
```

I belive when changing the background-color you will also likely need to change the grid color to something not white (assuming the background should be white).

tvt173, 2023-01-31:
nice. i wonder if there is also a way to set this globally in klayout programmatically, like you can with your user settings in the GUI... if not, might be good to add a way to set this with gdsfactory, similar to setting matplotlib flags, at the top of your script, i.e. `gf.visualization.set_background_color('white')`

sebastian-goeldi, 2023-01-31:
There is, but they are part of the MainWindow or even Application. These aren't available (at least as of 0.28.4) for the python package. What I would suggest is a Class that inherits from LayoutView and read a config from project/user/system

HWOde, 2023-02-01:
> There is, but they are part of the MainWindow or even Application. These aren't available (at least as of 0.28.3) for the python package. What I would suggest is a Class that inherits from LayoutView and read a config from project/user/system

HWOde, 2023-02-01:
@tvt173 @sebastian-goeldi  Sir, thank you for your kind reply, do you have any suggestion about some layers missing in the picture, besides the problem that the color of the picture in the document is black?

sebastian-goeldi, 2023-02-01:
I do not @HWOde . I believe this might be a configuration error (illegal layernumber/datatype), a duplicate (same numbers as a previous layer) or a bug in the way gdsfactory or gdstk handle it. Could you please share the code equivalent (the layer definitions) of what you used to produce the preview (assuming you are allowed to, which I assume since you shared the picture ;)).

If it is a bug @joamatab is probably the best person to handle it

HWOde, 2023-02-01:
> I do not @HWOde . I believe this might be a configuration error (illegal layernumber/datatype), a duplicate (same numbers as a previous layer) or a bug in the way gdsfactory or gdstk handle it. [...]

HWOde, 2023-02-01:
@sebastian-goeldi Sir, I rechecked my codes and confirmed they are ok, but I have some new discoveries, when I run the following codes 

c = LAYER_VIEWS.preview_layerset()
c.plot(blocking=True,show_subports=False)

I get the following image, and you can see that some layers are missing.

[image]

Then I run c.show() to pass the data to Klayout, and I was pleasantly surprised to find that in Klayout, the display of layers is correct, and no layers are missing.

[image]

So I think it should be a minor bug，actually when I run 

c = LAYER_VIEWS.preview_layerset()
c.plot(blocking=True,show_subports=True)  #change show_subports parameter

I got this picture

[image]

you can see all layers exists，after zooming in, you can see that the colors of those previously missing layers are transparent. This explains why there were missing layers in the previous image ,because I turned off the display of Port earlier, so the layer looks like it is missing.

[image]

sebastian-goeldi, 2023-02-01:
Ah, then it is probably a problem with the color palette for layer generation ;) i will defer that to @joamatab as I am not that familiar with the codebase

HWOde, 2023-02-01:
@sebastian-goeldi Sir, I agree with you, I think it may be because some colors generated by the LayerView object are not supported by matplotlib, so it does not know what color to display, so it set the color to transparent color or background color. Thanks again for your kind reply and help, have a nice day！

joamatab, 2023-02-03:
Hi HWOde,

if the visibility is False, in matplotlib we don't show the layers,

HWOde, 2023-02-03:
@joamatab Thank you for your kind reminder！

HWOde, 2023-02-03:
@joamatab Sir, I specify the visible property of all layers to be True, but matplot still fails to display some layers, e.g.

[image]    
[image]

[image]

Could you tell me where and how to set parameters to display all layers in matplotlib?

closed 2023-02-03 by joamatab

joamatab, 2023-02-03:
Hi HWOde,

i tried to explain it her

https://gdsfactory.github.io/gdsfactory/notebooks/03_layer_stack.html#LayerViews