gdsfactory/gdsfactory #1223: klayout webapp viewer running on your jupyter notebooks

joamatab, 2023-02-04:
Thanks to [Kweb](https://github.com/gdsfactory/kweb) you can now have klayout webapp viewer running on your jupyter notebooks

[image]

How can we get the kweb server running in the docs?
[image]

@proppy
@klayoutmatthias
@sebastian-goeldi
@tvt173

proppy, 2023-02-04:
I wonder if https://ipython.readthedocs.io/en/stable/config/integrating.html could help.

As Jupyter rendering could hint at a different representation type depending if it's meant for interactive usage or static rendering within the docs, and allow gdsfactory to choose between including an iframe or rendering a static image.

sebastian-goeldi, 2023-02-04:
This part is relatively easy. Declare jupyter_server as a global. I had it wrong in my mind about the declaration. Globals are module wide after reading about it, so it should be reasonably safe. Otherwise you have the problem that the setting of the server variable is not properly propagated between scopes.

What is left to do is, there are error dumps of websocket disconnects when you refresh cells. I haven't fully understood how to catch them without having the cell crash

joamatab, 2023-02-04:
Thank you Proppy and Sebastian

I was also surprised it does not work on binder either

[image]

closed 2023-10-24 by joamatab