gdsfactory/gdsfactory #1225: improve Ipython widget for klayout

joamatab, 2023-02-05:
How can we improve functionality for ipywidgets and ipyevents?

At the moment only zoom works 

[image]

it would be great to be able to move as well

tvt173, 2023-02-05:
functionality is in there but commented out... i think @sebastian-goeldi and i determined that it has something to do with the timer klayout is using which seems to need to be called continuously in the background

sebastian-goeldi, 2023-02-05:
https://github.com/KLayout/klayout/blob/master/src/layview/layview/layLayoutView_noqt.h#L67

;)

joamatab, 2023-02-20:
Thank you Thomas! @thomasdorch 

https://github.com/gdsfactory/gdsfactory/pull/1312

thomasdorch, 2023-02-20:
I haven't been able to get kweb to run to to see how it works yet but does dragging/scrolling work in it?

sebastian-goeldi, 2023-02-20:
Yes, in kweb all of this works. I mean dragging is not implemented. But you can scroll left right with [Ctrl]/[Alt]+MouseWheel and zoom with MouseWheel

joamatab, 2023-02-20:
yes, you can try this notebook

https://gdsfactory.github.io/gdsfactory/notebooks/workflow/jupyter_notebooks.html

tvt173, 2023-02-21:
i'm a bit unclear on all of this right now... what's the status on kweb vs the jupyter widget and when/how are we using one vs the other?

p.s. if we continue to use the widget, it would be nice to filter the layer list so that only those present in the layout are listed. otherwise the list is very long and unmanageable

joamatab, 2023-02-21:
kweb works well only when using one notebook, that's why is not used mainstream, and only used here
https://gdsfactory.github.io/gdsfactory/notebooks/workflow/jupyter_notebooks.html

I agree that list only present layers would be nice

tvt173, 2023-02-21:
i see... what's the issue when you have multiple notebooks?

sebastian-goeldi, 2023-02-21:
The problem isn't per se multiple notebooks. It's that it doesn't handle multiple instances yet. This shouldn't be hard, just need a small status that identifies it as a kweb and see make a call there to see whether kweb is running.

What I see more as a show stopper is that kweb currently will only run on localhost. This kills it if you were to run it on colab or any non-local jupyter(like). And there is no way ( to my knowledge) to get the url of a jupyternotebook.

The way that could work is with the jupyter proxy extensions. But that's where we are going to territories where it becomes heavy on users.

So, I would opt to bring the widget up to the features of kweb.

tvt173, 2023-02-21:
was just browsing the web for solutions for the kweb issue.... this one looks promising
https://colab.research.google.com/notebooks/snippets/advanced_outputs.ipynb#scrollTo=_7dYIo63EdgL

or this one for jupyterhub
https://docs.bokeh.org/en/latest/docs/user_guide/output/jupyter.html#jupyterhub

closed 2023-06-22 by joamatab