gdsfactory/gdsfactory #1227: data orchestration pipeline

joamatab, 2023-02-05:
For the data analysis pipeline it would be great to define the analysis of the data, from raw data to processed results as well as aggregated data

How could we build a data pipeline using a tool like dagster?

https://github.com/dagster-io/dagster

@mithro 
@poppy

mithro, 2023-02-06:
I think you mean @proppy

mithro, 2023-02-06:
Netflix released a thing called Papermill which looks like a pretty interesting option? See https://github.com/nteract/papermill

Apache Airflow is also a common option.

proppy, 2023-02-06:
See:
https://pyvideo.org/pydata-la-2019/data-and-etl-with-notebooks-in-papermill.html
https://pyvideo.org/pydata-la-2019/data-and-etl-with-notebooks-in-papermill.html

That's basically what we use with:
https://github.com/proppy/rad-lab-deploy/blob/tuning/rad-lab/modules/silicon_design/scripts/build/notebooks/examples/digital/serv/tuning.md

joamatab, 2023-08-15:
we moved dagster plugin into gplugins

https://github.com/gdsfactory/gplugins

closed 2023-08-15 by joamatab