gdsfactory/gdsfactory #1239: rename `gf.types`

joamatab, 2023-02-05:
gdsfactory.types can shadow `types` from the typechecker `import types` builtin module `types`

rename to gdsfactory.typings

tvt173, 2023-02-07:
`typings` is a bit of a weird name, no? `types` was good, but if that's problematic, `typing` is also fine. for example, that's how it's called in [numpy](https://numpy.org/doc/stable/reference/typing.html)

@joamatab

sebastian-goeldi, 2023-02-07:
`typing` has the same problem, it shadows the internal module while in the same folder. That's why the weird name. i don't know what else to use. A mypy extension such as `pylsp-mypy` complains about shadowing with either of the two

tvt173, 2023-02-07:
hmm... what exactly is the error we are getting? maybe it's a problem in the way we're running the tool? it seems odd that we can't use any of the good names, especially if other packages do 😄

sebastian-goeldi, 2023-02-07:
It's when used with a languageserver plugin and in the same folder as the `types.py `
[image]

It is not a big deal, but if we could avoid this, it would be nice.