gdsfactory/gdsfactory #1244: SiEPIC Verification errors

SebyLeDanger, 2023-02-06:
Dear Sir/Madam,

I am trying to layout MZIs using GDSfactory in VSCode.

Along the way, I met 2 obstacles that I have been unable to solve for now:

1. When I do the SiEPIC Verification in KLayout, I have an error saying that my grating couplers are not correctly oriented. I have the same problem by running an example of the ubcpdk library
2. When I add a floor plan, it is raising numerous error in KLayout all about the waveguides being overlapped. I have the same problem by running an example of the ubcpdk library

You can find my project here : https://github.com/SebyLeDanger/gdsfactory-mzi

PS: I also have two questions :

1. I cannot find an argument for the ubcpdk.add_fiber_array function to reduce the length of the horizontal waveguide to 0. We will also see in the gds file that the ubcpdk.add_fiber_array function create output waveguide of difference length for a reason that I ignore. I tried to use the gf.routing.add_fiber_array same issue
2. Is there a way to specify in python the EBeam technology so that I don't have to change it by hand in KLayout after running my code ?

joamatab, 2023-02-07:
Hi Seby,

i looked at your layout and it looks good to me

[image]

I think the issue is on the SiEPIC verification script

@lukas-ubc

SebyLeDanger, 2023-02-07:
Hi Joaquin,

Thank you for taking the time to look at my layout.

I know that the gratings are correctly oriented but there must be something off between gdsfactory and the Siepic verification. 

If you look at the properties of one grating, you will see that the rotation angle is not 0° but 270°. So I guess the reference position in gds is not the same than in the Siepic tool.

lukasc-ubc, 2023-02-07:
Hi folks,

The issue is that the grating coupler cell polygons were rotated when you created the PDK.  Please see the original cell in the PDK: https://github.com/SiEPIC/SiEPIC_EBeam_PDK/blob/master/klayout/EBeam/gds/EBeam/ebeam_gc_te1550.gds
Notice that it is facing to the right, whereas the cell in your PDK is facing left.

It is a really bad idea to modify foundry-provided cells. In this case we actually do an IP replacement for the grating couplers, so if you have rotated the contents of cell, after we replace it, the polygons will be pointing the opposite direction!

The definitions for verification Design for Test rules are specified in DFT.XML, as described here: https://github.com/SiEPIC/SiEPIC-Tools/wiki/SiEPIC-Tools-Menu-descriptions (search for DFT).

and for the EBeam PDK here:
https://github.com/SiEPIC/SiEPIC_EBeam_PDK/blob/master/klayout/EBeam/DFT.xml

Finally, note that we are now using Black Box versions for all the grating couplers, since different foundries have different fabrication process bias, and require calibration of the coupler for each process.  Then we perform an IP replacement just prior to fabrication.

Thank you
Lukas

SebyLeDanger, 2023-02-08:
Dear Lukas,

I am not exactly sure what you mean by :

> The issue is that the grating coupler cell polygons were rotated when you created the PDK

I have checked the ebeam_gc_te1550.gds file in the ubcpdk folder and it is pointing right with a 0° degree angle. So I am a bit confused. What could cause such a rotation when I create the pdk ?

Moreover, in the ubcpdk I could not find the Black Box versions of the GC components that I have to use. 

Thank you for taking the time to help me,
Sébastien

lukasc-ubc, 2023-02-08:
@joamatab showed me yesterday that this file was pointing left. 

The missing cells need to be imported into the GDSfactory implementation of the PDK.

SebyLeDanger, 2023-02-08:
Maybe it is because, when I open the ebeam_gc_te1550 from python it is pointing left but when I open the gds file from the windows explorer it is pointing right. Is there something rotating/mirroring the gc when importing the gds file in python ?

# Windows explorer
[image]

# Python
[image]

joamatab, 2023-02-08:
Hi Sebastian and Lukas,

I sent a PR that fixes the issue

as Lukas suggested we create a new container cell that has the original grating coupler rotated

[image]

joamatab, 2023-02-08:
If you `pip install ubcpdk --upgrade` your design works now

[image]

Here is a PR with the new grating coupler design

https://github.com/SebyLeDanger/gdsfactory-mzi/pull/1

closed 2023-02-08 by joamatab

SebyLeDanger, 2023-02-08:
Thanks a lot to both of you !

lukasc-ubc, 2023-02-16:
Hi Sebastien,

The original source PDK is here: https://github.com/SiEPIC/SiEPIC_EBeam_PDK

The PDK cells were imported by Joaquin and team at GDSfactory, into what they call “ubcpdk”.  

Making sure that ubcpdk is consistent with SiEPIC_EBeam_PDK is not automated. Presently they are independent and might not be identical.

I hope that clarifies.

thank you
Lukas

joamatab, 2023-02-16:
yes, Thomas made a script for that

https://github.com/gdsfactory/ubc/blob/main/update_source.sh