gdsfactory/gdsfactory #1252: get_bundle_all_angle: error in _centerpoint_offset_curve

95-martin-orion, 2023-02-08:
**Describe the bug**
With certain geometry, running `gf.routing.all_angle.get_bundle_all_angle` encounters the following error:

```
File "[REDACTED]/gdsfactory/path.py", line 265, in _centerpoint_offset_curve
    new_points[:, 0] -= offset_distance * np.cos(theta_mid)
ValueError: non-broadcastable output operand with shape (1,) doesn't match the broadcast shape (0,)
```

**To Reproduce**
Run the code sample below. The 11th wire is what fails; set `NUM_WIRES=10` at the top to render the first 10 wires.

<details><summary>Code to reproduce</summary>

```python
import gdsfactory as gf
from gdsfactory.routing.all_angle import get_bundle_all_angle

# Set to 10 for successful render
NUM_WIRES=11

@gf.cell
def inner_array():
    c = gf.Component()
    base = gf.components.straight(cross_section=gf.cross_section.strip).rotate(45)
    for x in range(10):
        for y in range(6):
            base_ref = c.add_ref(base).move((x*20 - 90, y*20 - 50))
            c.add_port(f"inner_{x}_{y}", port=base_ref.ports['o1'])
    return c

@gf.cell
def outer_array():
    c = gf.Component()
    base = gf.components.straight(cross_section=gf.cross_section.strip)
    for idx, theta in enumerate(range(0, 360, 6)):
        base_ref = c.add_ref(base).move((300, 0)).rotate(theta)
        c.add_port(f"outer_{idx}", port=base_ref.ports['o1'])
    return c

@gf.cell
def chip():
    c = gf.Component()
    inner = c << inner_array()
    outer = c << outer_array()
    routes = get_bundle_all_angle(
        ports1=inner.get_ports_list()[:NUM_WIRES],
        ports2=outer.get_ports_list()[:NUM_WIRES],
        cross_section=gf.cross_section.strip,
        bend=gf.components.bend_euler,
    )
    for route in routes:
        c.add(route.references)
    return c

if __name__ == "__main__":
    # c = inner_array()
    # c = outer_array()
    gf.get_active_pdk().register_cross_sections(strip=gf.cross_section.strip)
    c = chip()
    c.show(show_ports=True)
```

</details>

**Expected behavior**
Wires are routed from all ports of `inner_array` to all ports of `outer_array`. (Wire overlap isn't a concern in this issue - I'll file a separate bug for that.)

**Suggested fix**
It seems like there's an edge case in `gf.path._centerpoint_offset_curve` that causes `theta_mid` to be an empty numpy array. I don't have a good intuition for _why_ this happens, but it seems like the right place to start.

95-martin-orion, 2023-02-08:
CC @tvt173 - sorry to stress-test such a new feature!

tvt173, 2023-02-08:
thanks @95-martin-orion . i was porting some test cases over yesterday and also saw this same error for one case in the generic PDK which I did not see originally. have not looked deeper into it yet, but if you find the root of the error, please let me know! hoping to add a bigger swath of test cases and examples in the next few days

closed 2023-02-15 by joamatab