gdsfactory/gdsfactory #1253: get_bundle_all_angle: overlapping wires

95-martin-orion, 2023-02-08:
**Describe the bug**
`get_bundle_all_angle` will generate overlapping wires for certain geometry.

**To Reproduce**
Run the code sample below.

<details><summary>Code to reproduce</summary>

```python
import gdsfactory as gf
from gdsfactory.routing.all_angle import get_bundle_all_angle

NUM_WIRES=10

@gf.cell
def inner_array():
    c = gf.Component()
    base = gf.components.straight(cross_section=gf.cross_section.strip).rotate(45)
    for x in range(10):
        for y in range(6):
            base_ref = c.add_ref(base).move((x*20 - 90, y*20 - 50))
            c.add_port(f"inner_{x}_{y}", port=base_ref.ports['o1'])
    return c

@gf.cell
def outer_array():
    c = gf.Component()
    base = gf.components.straight(cross_section=gf.cross_section.strip)
    for idx, theta in enumerate(range(0, 360, 6)):
        base_ref = c.add_ref(base).move((300, 0)).rotate(theta)
        c.add_port(f"outer_{idx}", port=base_ref.ports['o1'])
    return c

@gf.cell
def chip():
    c = gf.Component()
    inner = c << inner_array()
    outer = c << outer_array()
    routes = get_bundle_all_angle(
        ports1=inner.get_ports_list()[:NUM_WIRES],
        ports2=outer.get_ports_list()[:NUM_WIRES],
        cross_section=gf.cross_section.strip,
        bend=gf.components.bend_euler,
    )
    for route in routes:
        c.add(route.references)
    return c

if __name__ == "__main__":
    # c = inner_array()
    # c = outer_array()
    gf.get_active_pdk().register_cross_sections(strip=gf.cross_section.strip)
    c = chip()
    c.show(show_ports=True)
```

</details>

**Expected behavior**
Wires generated by the same `get_bundle_all_angle` call do not overlap.

**Suggested fix**
This may be a symptom of me misusing a very new feature, rather than an actual bug. If so, documentation on how to use this tool appropriately would be sufficient to close this issue.

If this is an actual bug (i.e. wires are not expected to overlap), maybe have wires run in parallel as long as possible, then fanout? This seems like a particularly complex problem, though.

95-martin-orion, 2023-02-08:
CC @tvt173 for all-angle routing.

tvt173, 2023-02-09:
wow, quite an interesting example you give... adding a snapshot for reference...
[image]

this is currently a limitation when you have the case of a "direct" connection like this, with no steps defined. essentially bundling logic is ignored. this case is especially tricky, because the input/output ports are not facing each other, so the route needs to do an around-the-back S to acheive the connection. to really properly bundle in this case, the outer routes would need to either use a 180 euler with larger bend radii or insert a straight between two 90(ish) bends... that is also not supported right now. so, in this case, each route essentially does its best on its own, without consideration for the bundle. my suggestion to fix it, at this point, would be to bundle by hand, with custom steps, like shown below:

```python
import gdsfactory as gf
from gdsfactory.routing.all_angle import get_bundle_all_angle

NUM_WIRES=10

@gf.cell
def inner_array():
    c = gf.Component()
    base = gf.components.straight(cross_section=gf.cross_section.strip).rotate(45)
    for x in range(10):
        for y in range(6):
            base_ref = c.add_ref(base).move((x*20 - 90, y*20 - 50))
            c.add_port(f"inner_{x}_{y}", port=base_ref.ports['o1'])
    return c

@gf.cell
def outer_array():
    c = gf.Component()
    base = gf.components.straight(cross_section=gf.cross_section.strip)
    for idx, theta in enumerate(range(0, 360, 6)):
        base_ref = c.add_ref(base).move((300, 0)).rotate(theta)
        c.add_port(f"outer_{idx}", port=base_ref.ports['o1'])
    return c

@gf.cell
def chip():
    c = gf.Component()
    inner = c << inner_array()
    outer = c << outer_array()
    inner_ports = inner.get_ports_list()
    outer_ports = outer.get_ports_list()
    for n_route in range(NUM_WIRES):
        routes = get_bundle_all_angle(
            ports1=[inner_ports[n_route]],
            ports2=[outer_ports[n_route]],
            cross_section=gf.cross_section.strip,
            bend=gf.components.bend_euler,
            start_angle=-40,
            steps=[{'ds': (NUM_WIRES - n_route) * 20},]
        )
        for route in routes:
            c.add(route.references)
    return c

if __name__ == "__main__":
    # c = inner_array()
    # c = outer_array()
    gf.get_active_pdk().register_cross_sections(strip=gf.cross_section.strip)
    c = chip()
    c.show(show_ports=True)
```

which gives
[image]

95-martin-orion, 2023-02-09:
> wow, quite an interesting example you give {...}

Just a quick-and-dirty toy design made for the sole purpose of isolating this and #1252 :slightly_smiling_face: 

> {...} my suggestion to fix it, at this point, would be to bundle by hand {...example...}

Thanks, this example is great! It sounds like the root issue here was my design not providing sufficient detail to `get_bundle_all_angle` rather than an actual bug. Assuming that's the case, my only remaining request on this issue is to include a note about this in the eventual docs for this feature.

joamatab, 2023-02-09:
sounds good,

we can track the improving the non-manhattan router docs in this issue
https://github.com/gdsfactory/gdsfactory/issues/1255

closed 2023-09-12 by joamatab