gdsfactory/gdsfactory #1266: define test sequence and data analysis flow

joamatab, 2023-02-12:
At the moment we have functions to easily add abstract GDS [labels](https://gdsfactory.github.io/gdsfactory/notebooks/07_mask.html) in the GDS file

we can also `Component.write_gds_with_metadata()` to store all metadata in YAML format

How could we define the test sequence and data analysis for each cell?

The way we have been doing it now is to use `Component.info` to add tags for experiments. The info gets saved into YAML thanks to  `Component.write_gds_with_metadata()` 

For example for a heated_ring filter

```
import gdsfactory as gf

@gf.cell
def wg():
    c = gf.Component()
    c.info['doe'] = ['rings', 1550, 'te', 'phase_shifter']
    c.info['test_sequence'] = ['optical', 'electrical_sweep']
    c.info['data_analysis'] = ['remove_baseline', 'extract_fsr', 'extract_loss', 'extract_power_per_pi']
    return c

c = wg()
c.pprint()
c.write_gds_with_metadata('a.gds')
````

It would be great to come up with a standard together with SiEPIC, so we can also use it for the edx course
https://github.com/gdsfactory/ubc/issues/176

Ideally we would use an intermediate representation such as YAML

@HelgeGehring 
@simbilod 
@mdecea 
@SkandanC
@nikosavola
@jolzgrafe
@lukasc-ubc
@JonathanCauchon
@heitzmann
@tvt173 
@Jan-David-Black 
@afischer-sweepme 
@sequoiap 
@bilderbuchi

lukasc-ubc, 2023-02-12:
Hi Axel,

Our starting point was a flat list, where you can specify arbitrary pairs of voltages for the two channels.  We thought about the tree, but then it quickly leads to NxM measurements.  We thought that people can run their own nested forloops and create the flat measurements. that way it is easier to get a sense of how long the measurements will take.

The other complexity I see that we don't have a solution for is doing a measurement, analysing the data, then doing another measurement.  e.g., sweep heater 1, find the maximum transmission point, then bias heater 1, then sweep heater 2.

afischer-sweepme, 2023-02-13:
By adding further evaluation steps, the test sequence will become a program itself that needs to be interpreted on a higher level than just providing a list of measurement points that can easily be run through by a script. It will probably lead to the situation that you need to develop and maintain a own measurement software to run and interprete your test sequence files which in turn needs to be quite flexible to be adapted to the setups of different users. This will easily become a larger project on the level of what we do.

@joamatab Let's make a video meeting. I can demonstrate how we tackle things although our system also not has answers to all questions. Still, it might give you an idea of the problems that we already faced.

In our approach, we have human-readable setting files that are generated by combining modules in a GUI sequencer tool. Our settings represent the entire program. Still,  it would be possible to overwrite placeholder parameters in these files in order to customize a pre-defined program procedure to new measurement ranges e.g. a variation of start and end wavelength scan of a spectral scan. It means that gdsfactory could use a setting file for a certain procedure and customize it to the needs of the chip to be measured. Still one would need for each measurement an own setting file and for each setting file one would need to define how parameters are replaced. Here, we could probably improve our software if there would be setup- and test sequence-specific configuration files that can overwrite properties of the base setting files so that each user can customize the setting to a setup and gdsfactory only needs to write the test sequence-specific parameters.

bilderbuchi, 2023-02-13:
I think the answer to "How could we define the test sequence and data analysis for each cell?" needs first some clarity what kind of logic you want to implement.
* On the low end of the complexity scale, there is a list of commands in a text file (e.g. yaml).
* On the opposite end, there is a full programming language - you could use Python code directly.
* Somewhere in-between, there is using a custom DSL like https://gitlab.com/librecube/prototypes/python-pluto (used in an aerospace context, [poster on this}(https://events.libre.space/event/3/contributions/49/attachments/45/66/librecube_pluto.pdf)). Here you can implement as much complexity as you need, and tune the capability-complexity tradeoff to your needs. Depending on that, it might be a lot of work, though.

tvt173, 2023-02-13:
hi @joamatab ,

as discussed before, I recommend requiring that blocks to be tested have ports exposed at the top level. then you can have a decorator which scans the top level circuit(s) and generates test metadata based on the globally-unique port names + info included on the attached subcomponents/circuits.

@bilderbuchi ,

interesting example you give with pluto... i don't see why it would be advantageous over a yaml file with a clear schema and appropriate parser though. i would prefer to not introduce obscure file formats unless necessary, and I think that in most cases, code generation ends up being more unwieldy and less maintainable than a schema/parser flow

lukasc-ubc, 2023-02-14:
Thank you for your thoughts folks. Axel, is your project being open sourced? We have certainly thought about a graphical sequencer but that is indeed a bigger endeavour. 

Sounds like a list of measurements like we are doing is a reasonable compromise. 

It would be nice to see what you are doing.

afischer-sweepme, 2023-02-14:
@lukasc-ubc, all our instrument drivers are open-source and we published [pysweepme](https://github.com/SweepMe/pysweepme), a library to load our instrument drivers in own python projects, both under MIT license. I do not know which driver scheme you are using so far but you could consider to use an existing one to benefit from others that already wrapped communication commands etc. 

SweepMe! itself is not open-source as we started to sell lifetime licenses for a bundle of add-on modules as one part to finance our business. The main software can be used for free and everybody is free to add drivers on its own. With pysweepme we have started to outsource code from SweepMe! and we consider to "open outsource" further parts of SweepMe!. However, these developments have do not have the highest priority as we primarily need to concentrate on customer projects to finance us. In these projects, we then often create instrument drivers that are afterwards made available again open-source to everybody.

joamatab, 2023-10-24:
Marked this as complete, 

See docs here
https://gdsfactory.github.io/gdsfactory/notebooks/07_mask.html#metadata

closed 2023-10-24 by joamatab