gdsfactory/gdsfactory #1271: KLayout connectivity problem with Windows

vangapandu95, 2023-02-13:
**Describe the bug**
What's the bug?

**To Reproduce**
What's the code to reproduce the behavior? What commands or code did you write to get the error?
You can add screenshots to help explain your problem.
Adding the code in `fixme/` folder is also a great option for others to reproduce your issue.

**Expected behavior**
What would you like to happen?

**Suggested fix**
How could we fix the bug?

vangapandu95, 2023-02-13:
I'd like to know if anyone has experience working on VScode and KLayout in a windows system.
I open the VScode through anaconda navigator and the interpreter path is set to the .exe file in the conda folder. This does not raise any issues with the interpreter. The code runs without error, but I dont see anything showing up in the KLayout window. Had anyone faced this issue ? I have the packages of GDSFactory installed in the KLayout.
[image]
[image]
I am stuck at this point, so any help is appreciated.

tvt173, 2023-02-14:
hi @vangapandu95, i briefly had this issue, and then it seemed to resolve itself (I believe with an upgraded version of gdsfactory?). at that point, I had to go into the gdsfactory menu bar, as shown in your snapshot and click an option like "enable klive server" or something to that effect.
[image]

i think that in more recent versions of gdsfactory, both that option and issue go away

joamatab, 2023-02-14:
Hi @vangapandu95, 

make sure you restart klayout after install the klayout integration

vangapandu95, 2023-02-14:
Thanks very much @tvt173 , and @joamatab !!
I appreciate it. I uninstalled my older version and just used _pip install_ in anaconda to install the latest version. The KLayout now shows the layouts live after running the scripts in VScode

closed 2023-02-14 by joamatab