gdsfactory/gdsfactory #1278: Multiprocessing in gdsfactory / Tidy3D

tylerflex, 2023-02-15:
We are considering integrating a tool like `multiprocessing` into tidy3d for our batch handling.

However, I'm having a lot of issues getting it working smoothly on MacOS and it does not seem like a reliable option.

Since our tools are linked through the tidy3d plugin, do you have any suggestions for other multi-processing tools (or queuing tools) that we could use in our web.Batch and still maintain good compatibility with gdsfactory?

@momchil-flex 
@joamatab

joamatab, 2023-02-15:
Hi Tyler,

I liked the first version of the plugin where we used threads more than the one we have  now

https://github.com/gdsfactory/gdsfactory/commit/4e56be623d69c705881f55bdaaaff6b58d9630fb 

see 5.10.3 version before we removed the threads

https://github.com/gdsfactory/gdsfactory/tree/v5.10.3

tvt173, 2023-02-15:
i always recommend [dask](https://www.dask.org/get-started)... it's powerful, simple to use, and works with multithreading, multiprocessing, and even distributed backends. that said, you should always be careful when using gdsfactory with multiprocessing. i imagine that for your needs, you shouldn't _need_ to create cells on worker processes, so I wouldn't. create all cells in the main process, then pass the required info to the worker processes where you are doing your tidy3d stuff.

tylerflex, 2023-02-15:
Thanks @tvt173 , I'll check out dask in more detail. We already have it as a requirement due to xarray so it might be a good option.

simbilod, 2023-02-16:
Another option: I've been replacing multiprocessing by [ray](https://docs.ray.io/en/latest/ray-more-libs/multiprocessing.html) multiprocessing, it works well

tylerflex, 2023-02-23:
see discussion in #1327 , I ended up going with `asyncio`, not perfect but was the only approach I could get working after trying a bunch of stuff.

closed 2023-02-23 by tylerflex