gdsfactory/gdsfactory #1282: secret bug in gf.cell

gdspaul, 2023-02-16:
In the code quoted below from `cell.py`, I know this never happens, but if you DO pass any args or kwargs to `cell_without_validator`, it will crash.  The documentation for cell() indicates that it accepts a number of keyword arguments.

Purpose: I'm interested in manipulating the caching behavior of cells; I implemented a decorator that looks for a FILE cache of the component gds and yml and loads that.  This is actually very handy for saving time assembling large layouts, but gdsfactory caching gets confused by my decorator.

```
def cell(func: _F, *args, **kwargs) -> _F:
    # ...
    return cell_without_validator(validate_arguments(func), *args, **kwargs)
```

joamatab, 2023-02-16:
Hi Paul, can you share the code to reproduce?

tvt173, 2023-02-16:
agreed that this is a bug and improper documentation of the `cell` decorator function. kwargs which are documented are not actually kwargs to the decorator itself, but they are injected into the signature of the wrapped function.

i.e. this fails
```python
@cell(cache=False)
def my_cell():
   ...
```

but this works
```python
my_cell(cache=False)
```

I believe this is the intended behaviour, but the docs are misleading
[image]

it should say that these kwargs are injected into the resulting functions. they are not actually kwargs to the decorator itself

gdspaul, 2023-02-16:
Exactly @tvt173 .  The documentation is wrong, but also _technically_ the code is wrong because if args or kwargs is **non-empty** then it will crash.  Like I said, it's almost impossible to make this happen.  But updating the docs would be nice.  Thanks guys.  :-)

joamatab, 2023-02-16:
yes, 

Troy already mentioned that the decorator docs were wrong

let me know if this makes it more clear
https://github.com/gdsfactory/gdsfactory/pull/1287

closed 2023-02-16 by joamatab

gdspaul, 2023-02-16:
My favorite part:
[image]

Impeccable service as always, Mr. Matres!