gdsfactory/gdsfactory #1289: Wait to load LayerViews until needed

thomasdorch, 2023-02-16:
Currently, the generic LayerViews is loaded when the gdsfactory module is imported, which doesn't make sense if we are loading a particular PDK right after that, which will load its own LayerViews. 

Instead, we should wait to load LayerViews until something needs to be displayed.

joamatab, 2023-02-18:
good idea,
at the moment we need to activate the generic PDK, so he LayerViews get activated

We also have some issues when loading complicated lyp files
https://github.com/gdsfactory/skywater130/issues/59

@tvt173
@skychil

tvt173, 2023-02-18:
_long term_, it would be good to converge on how we want pdk's to be activated in general... we discussed many options, such as adding a config file in our projects to specify which PDK to use. 

but _short term_, couldn't we just move this to `get_active_pdk()`... if there is no active PDK yet, load the generic PDK? i think everything should still work and it would solve this problem

closed 2023-02-19 by joamatab