gdsfactory/gdsfactory #1293: Circuit (yaml) parser type by file extension

tvt173, 2023-02-17:
**Is your feature request related to a problem? Please describe.**
We now allow the Pdk creator to specify which function to use as their `circuit_yaml_parser`. But it might be a nice idea to have this be a dictionary of functions keyed by file extension. Then, for example, we could map `*.pic.yml` to the traditional yaml parser by default and `*.pic.yml.jinja2` to the jinja-yaml parser. That would allow us to support both formats within the same project and allow most everyone to stay with default settings. As a bonus, IDEs like VSCode can parse `*.yml.jinja2` correctly (with the proper extension) without meddling deep into the settings and messing up yaml comprehension for other normal yaml files.

**Describe the solution you'd like**
Today (in generic PDK):
```python
circuit_yaml_parser: Callable = cell_from_yaml
```
Proposed:
```python
circuit_yaml_parsers: Dict[str, Callable] = { '*.pic.yml': cell_from_yaml, '*.pic.yml.jinja2': cell_from_yaml_template}
```

joamatab, 2023-02-17:
sounds good to me Troy

joamatab, 2023-02-19:
Maybe we can keep only one type, Im happy adopting the new Jinja2 as it seems an extension of the one we had.

@Jan-David-Black

tvt173, 2023-02-20:
well, it would certainly be easier to only have to support one format. if no one objects to it, it should be fairly easy to deprecate nicely also... you would only need to check if the `settings` entry is present in the yaml file... that's the only uncompatible difference. so you could throw a deprecation warning if `settings` is present and use the old parser function in that case, and we could remove support entirely in a future version

i think the only downsides to the jinja format are
- the files are no longer pure yaml
- they can start to become fairly complex

but overall i prefer it as it can be much more powerful for complex circuits