gdsfactory/gdsfactory #1298: DEVSIM TCAD simulator example not working

dominauta, 2023-02-18:
**Describe the bug**

When running the example code in https://gdsfactory.github.io/gdsfactory/notebooks/plugins/devsim/01_pin_waveguide.html?highlight=devsim, the mode solver fails to run.

<details><summary>Python terminal error output</summary>

```python

File "C:\Users\david\AppData\Local\Temp\ipykernel_21052\2819143626.py", line 1, in <module>
    waveguide.compute_modes(
File "C:\ProgramData\Anaconda3\lib\site-packages\gdsfactory\simulation\gtidy3d\modes.py", line 409, in compute_modes
    nx = self.get_n(
File "C:\ProgramData\Anaconda3\lib\site-packages\gdsfactory\simulation\gtidy3d\modes.py", line 270, in get_n
    ncore = self.get_ncore()
File "C:\ProgramData\Anaconda3\lib\site-packages\gdsfactory\simulation\gtidy3d\modes.py", line 255, in get_ncore
    return get_material_index(self.ncore, wavelength)
File "C:\ProgramData\Anaconda3\lib\site-packages\gdsfactory\pdk.py", line 545, in get_material_index
    return _ACTIVE_PDK.get_material_index(material, *args, **kwargs)
File "C:\ProgramData\Anaconda3\lib\site-packages\gdsfactory\pdk.py", line 514, in get_material_index
    raise ValueError(f"{key!r} not in {material_names}")
ValueError: functools.partial(<function get_index at 0x000001A1E9DC41F0>, 'si') not in ['si', 'sio2', 'sin'] 
```
</details>

**To Reproduce**
<details><summary>Code to reproduce the error</summary>

```python

import numpy as np
import matplotlib.pyplot as plt
from gdsfactory.simulation.devsim import get_simulation_xsection
from gdsfactory.simulation.devsim.get_simulation_xsection import k_to_alpha

nm = 1e-9
c = get_simulation_xsection.PINWaveguide(
    wg_width=500 * nm,
    wg_thickness=220 * nm,
    slab_thickness=90 * nm,
)
c.ddsolver()
n_dist = {}
neffs = {}
for ind, voltage in enumerate(voltages):
    if ind == 0:
        Vinit = 0
    else:
        Vinit = voltages[ind - 1]

    c.ramp_voltage(Vfinal=voltage, Vstep=ramp_rate, Vinit=Vinit)
    waveguide = c.make_waveguide(wavelength=1.55)
    waveguide.compute_modes(
        isolate=True
    )
```
</details>

**Expected behavior**
The code should run the mode solver with a perturbed waveguide

**Suggested fix**
The parameters 'ncore' and 'nclad' from the object waveguide are expecting a string like 'si' or 'sio2' and not a functools.partial object.

Adding the following lines before the call to waveguide.compute_modes() makes the example run

```python
waveguide.ncore='si'
waveguide.nclad='sio2'
```
By changing the gdsfactory/simulation/gtidy3d/modes.py file like this is more elegant (I'm not sure if this is a proper fix or any other code part will stop working):

Lines 253 to 259:
```python
    def get_ncore(self, wavelength: Optional[float] = None) -> float:
        wavelength = wavelength or self.wavelength
        return get_material_index(self.ncore.args[0], wavelength)

    def get_nclad(self, wavelength: Optional[float] = None) -> float:
        wavelength = wavelength or self.wavelength
        return get_material_index(self.nclad.args[0], wavelength)
```

joamatab, 2023-02-18:
great catch David

Im updating the default Waveguide modeSolver mode in devsim to have a default ncore='si' and nclad='sio2'

```
        ncore: core refractive index material spec. Can be:
            string: material name.
            float: refractive index.
            float, float: refractive index real and imaginary part.
            function: function of wavelength.
```

Im also adding a test for it

simbilod, 2023-02-19:
It seems caching was changed as well, by deactivating caching of tidy3d modes I can get the modulation back:

[image]

simbilod, 2023-02-19:
see #1303

joamatab, 2023-02-19:
fixed it https://github.com/gdsfactory/gdsfactory/pull/1304

closed 2023-02-20 by joamatab