gdsfactory/gdsfactory #1300: reduce size of git repo

joamatab, 2023-02-19:
How can we reduce the size of the GIT repo?

At some point we commited some unnecessary data

sebastian-goeldi, 2023-02-19:
Most often I have seen this one mentioned for this use: https://github.com/rtyley/bfg-repo-cleaner

tvt173, 2023-02-20:
yes, i've used BFG a few times for this... works like a charm. you'll need to do it locally once then force-push the repo. this could be disruptive to those who pull later though just to warn (like any operation re-writing git history)

joamatab, 2023-02-20:
I tried and i got this

https://github.com/rtyley/bfg-repo-cleaner/issues/36

[image]

sebastian-goeldi, 2023-02-20:
You have to force push, otherwise it worn't work, and also you might need to temporarily remove protections agains force push

joamatab, 2023-02-20:
`git push -f` did not work either

`git remote update` worked

https://christoph.ruegg.name/blog/git-howto-mirror-a-github-repository-without-pull-refs.html

I think the repo went down from 278Mb to 14MB, Im going to delete some more files, such as old images

joamatab, 2023-02-20:
the repo is 11Mb now :tada: 

[image]

joamatab, 2023-02-23:
for some reason the repo size went back to 117MB

[image]

joamatab, 2023-02-23:
it should be fixed again by moving docs not in a branch

closed 2023-02-23 by joamatab