gdsfactory/gdsfactory #1310: add_ref(x_reflection) not working

joamatab, 2023-02-20:
How can we fix the gdstk.Reference(x_reflection)?

```
c = gf.Component('demo')
c.add_ref(gf.components.C(), x_reflection=True)
c.show()
```

[image]

@heitzmann

heitzmann, 2023-02-20:
@joamatab The argument `x_reflection` mirrors across the x axis, equivalent to gdsfactory's `mirror((0, 0), (1, 0))`, so there's no bug in the gdstk implementation:

```python
import gdstk

c0 = gdstk.Cell("0").add(*gdstk.text("F", 4, (0, 0)))

c1 = gdstk.Cell("1").add(
        gdstk.Reference(c0),
        gdstk.Reference(c0, origin=(5, 0), x_reflection=True)
    )

c1.write_svg("x_reflection.svg")
```
[image]

joamatab, 2023-02-20:
Thank you Lucas

you are right, it's correct :)

```
import gdsfactory as gf

c = gf.Component('demo')
r1 = c.add_ref(gf.components.C(), x_reflection=True)
r2 = c.add_ref(gf.components.C(), x_reflection=False)
r2.movex(20)
c
```

[image]

closed 2023-02-20 by joamatab