gdsfactory/gdsfactory #1327: run multiple web simulations in paralell

joamatab, 2023-02-22:
How could we run multiple simulations in paralell using tidy3d?

# Using threads

One option is to break the write_sparameters function into 2 parts:

1. submits the job and returns an awaitable result
2. process the result

# Using Ray

#1278 

@lucas-flexcompute 
@tyler-flex
@simbilod 
@tvt173 
@flaport 
@nikosavola

tylerflex, 2023-02-23:
We introduced a `tidy3d.web.run_async` function, which will be released in the official 1.9.0 coming soon.

This uses asyncio to do concurrent tidy3d simulations, instead of needing `tidy3d.web.Batch`. It accepts a dictionary mapping of task_name -> Simulation and returns a `BatchData`, which can be iterated over like a dictionary.

I tried a few other approaches, including dask, but this one ultimately was what worked.

There's also an `tidy3d.plugins.adjoint.web.run_async`, which lets one run several simulations concurrently and still have the gradients tracked by jax. Notebook example coming out soon for that as well.

Maybe this functionality can be wrapped by gdsfactory?

tylerflex, 2023-02-23:
Note also, we will have a `verbose: bool` option in all web-related functions if you want to suppress our output and wrap tidy3d with your own progessbars / output

joamatab, 2023-02-23:
nice, being able to lower verbosity sounds great

ianwilliamson, 2023-03-05:
If the use case is to submit multiple simulation requests without blocking on the client side, then Python [futures](https://docs.python.org/3/library/concurrent.futures.html) is likely the most reasonable option. Specifically, use a `ThreadPoolExecutor` to submit multiple requests and then block on their `result()`. You do not need queues, asyncio, or dask (#1327, #1278).

joamatab, 2023-03-05:
Hi Ian, yes, that was the original implementation 

Momchil had to disable because it was messing up with tidy3d progress bars

I would like to bring it back

joamatab, 2023-04-21:
fixed by https://github.com/gdsfactory/gdsfactory/pull/1541

closed 2023-04-21 by joamatab