gdsfactory/gdsfactory #1330: documentation not building

joamatab, 2023-02-22:
Documentation is taking long time to build

it usually takes around 23 minutes and now it's not finishing in the max  6 hour limit from GitHub

in the `docs/conf.py` we are having 60 seconds timeout 

https://github.com/gdsfactory/gdsfactory/actions/workflows/pages.yml

@simbilod

joamatab, 2023-02-22:
I think is coming from the model extraction notebook, where we use `%capture` magic

i noticed that capture has the side effect of errors are not properly catched

Maybe also does not get affected by timeout?

Im running the notebook `02_model_extraction.ipynb` locally and pushing the data to gdslib, so we can leverage the cache to get rid of the timeout

joamatab, 2023-02-22:
Removed `%%capture` from suspected notebooks

https://github.com/gdsfactory/gdsfactory/pull/1328

simbilod, 2023-02-22:
oof I see. These notebooks have a lot of computation because they loop over inputs, maybe we should deactivate them for now?

joamatab, 2023-02-23:
The issue is coming from the ipython widget,

it makes it run of memory and kernel dies

@thomasdorch

joamatab, 2023-02-23:
i disabled the execution until we figure out how to fix this

closed 2023-02-24 by joamatab