gdsfactory/gdsfactory #1342: have ipynb and py versions of the notebooks under /docs

simbilod, 2023-02-23:
@joamatab is it possible to also generate the ipynb notebooks in the docs?

Right now, when a user clones the repo, they only get the .py version of the documentation and have to compile the notebook with jupytext themselves

It would be great if when cloning the repo you can also just run notebooks

joamatab, 2023-02-23:
yes,

see #1340 
there is a line on the `Makefile` that can  `make notebook`  as well as `make jupytext` in case someone wants to keep using notebooks, 

`make notebook` makes the notebooks
`make jupytext` convert notebooks to python in case you modified the notebooks

I was going to recommend using https://marketplace.visualstudio.com/items?itemName=donjayamanne.vscode-jupytext
to open the python files

simbilod, 2023-02-23:
My use case was a little different, as I was having people log in into a JupyterLab custer with a preset environment, clone the repo and then directly run notebooks

I guess I can install jupytext and have them compile the notebooks, or compile them myself somewhere for them to copy

joamatab, 2023-02-23:
there is also a jupyterlab jupytext extension that will also allow you to edit directly the py file

simbilod, 2023-02-24:
It was painful but I got people to run make notebooks :)

closed 2023-02-24 by simbilod

joamatab, 2023-02-25:
Is there not extension for jupytext  on jupyter lab? 

@proppy

tvt173, 2023-02-25:
Yes, in fact, just pip installing jupytext should be enough, according to
the docs

joamatab, 2023-02-27:
yes, it does

[image]