gdsfactory/gdsfactory #1346: notebook images failing

joamatab, 2023-02-24:
since we switch to jupyterbook some notebooks are failing to build correctly

TODO:

- [ ] check plugins

https://github.com/gdsfactory/gdsfactory/pull/1345

joamatab, 2023-02-24:
I also tried using plotly  but I get `Stream has unknown format`

```
            gdspath = component.write_gds(logging=False)
            lyp_path = gdspath.with_suffix(".lyp")

            layer_views = get_layer_views()
            layer_views.to_lyp(filepath=lyp_path)

            layout_view = lay.LayoutView()
            layout_view.load_layout(str(gdspath))
            layout_view.max_hier()
            layout_view.load_layer_props(str(lyp_path))

            import plotly.express as px

            img = layout_view.get_pixels(800, 600)
            fig = px.imshow(img)
            fig.show()

```

sebastian-goeldi, 2023-02-24:
have you tried with png_data instead of the pixels? this gives you only a pixelbuffer afaik.

joamatab, 2023-02-24:
yes, the issue is coming from `layout_view.load_layout(str(gdspath))` not sure why it does not work

klayoutmatthias, 2023-02-24:
I think that again is the plugin registration. 

The stream readers are packed in separate shared objects which are looked up relative to the "db" library. If there is no such stream reader plugin lib, the load fails with "stream has unknown format".

Matthias

klayoutmatthias, 2023-02-24:
Here is a sane file structure from a python module build of mine (shortened):

```
klayout/   <-- module root
├── db
│   ├── __init__.py
│   └── pcell_declaration_helper.py
├── dbcore.cpython-310-x86_64-linux-gnu.so   <-- dbcore python import library
├── db_plugins
│   ├── lib_cif_dbpi.cpython-310-x86_64-linux-gnu.so    <-- stream readers
│   ├── lib_dxf_dbpi.cpython-310-x86_64-linux-gnu.so
│   ├── lib_gds2_dbpi.cpython-310-x86_64-linux-gnu.so
│   ├── lib_lefdef_dbpi.cpython-310-x86_64-linux-gnu.so
│   ├── lib_magic_dbpi.cpython-310-x86_64-linux-gnu.so
│   ├── lib_mebes_dbpi.cpython-310-x86_64-linux-gnu.so
│   ├── lib_net_tracer_dbpi.cpython-310-x86_64-linux-gnu.so
│   ├── lib_oasis_dbpi.cpython-310-x86_64-linux-gnu.so
│   └── lib_pcb_dbpi.cpython-310-x86_64-linux-gnu.so
├── lib_db.cpython-310-x86_64-linux-gnu.so   <-- KLayout kernel, db classes
```

So "db_plugin" needs to be next to the `lib_db.cpython-310-x86_64-linux-gnu.so` library and needs to be filled with the stream reader plugin so's.

If you have this structure and it still does not work, LD_DEBUG is your friend.

Matthias

proppy, 2023-02-24:
did you make sure to import:
```
import klayout.db
```
before importing:
```
import klayout.lay
```
see:
https://github.com/KLayout/klayout/issues/1146#issuecomment-1345643295

joamatab, 2023-02-24:
That worked great! thank you proppy!

I also think i identified the issue with the widget,
https://github.com/gdsfactory/gdsfactory/commit/714c6f142e3aaba7c9fbc1e227aa277e948fd818

And Im sending a PR of a version that uses a static image instead of a widget

joamatab, 2023-02-24:
it's fixed, we just needed to pin to an older version of ipywidgets

https://gdsfactory.github.io/gdsfactory/notebooks/00_geometry.html

Sending a PR that make ti even better

closed 2023-02-24 by joamatab

klayoutmatthias, 2023-02-24:
Ah, well ... sorry, did not thing about "import klayout.db" ... but you're right. That will load the stream readers.

proppy, 2023-02-25:
@klayoutmatthias would it make sense to do it systematically in `klayout.lay` `__init__.py`? If yes, I'd be  happy to file an issue and start working on it!

klayoutmatthias, 2023-02-25:
Yes, that definitely makes sense!

I have included that already in the wip branch (https://github.com/KLayout/klayout/pull/1301/files/a3ff671eb9b998b17f5474b29bb8ad437e93c0c9..04626f9c2beff9750edf45cda6da4e6de6e78953). It will go into 0.28.6.

Matthias