gdsfactory/gdsfactory #1353: Running FDTD Lumerical simulation on a component

vangapandu95, 2023-02-25:
I am exploring the GDSFactory's capabilities especially to interface it with Lumerical FDTD
While referring to the plugin content listed on "https://gdsfactory.github.io/gdsfactory/notebooks/lumerical/1_fdtd_sparameters.html", I got the impression that by writing the line of a code you can create an FDTD layout in Lumerical and run the simulations and get the data written all in single step without having to import the GDS file into an FDTD window and placing the ports manually and run the FDTD. Please correct me if this is not correct.
For example, I would like to run FDTD simulation of the bezier bend. In the figure shown below, I get the layout opened in the KLayout window when bezier.py script is run in VSCode. Can I simulate this bezier curve in FDTD lumerical on my computer by adding more code to the bezier.py file? Is there any reference that'd help me how to proceed?
[image]

joamatab, 2023-02-25:
Yes, here is the tutorial on going from layout to sparameters 

https://gdsfactory.github.io/gdsfactory/notebooks/lumerical/1_fdtd_sparameters.html

vangapandu95, 2023-02-25:
Thanks Joaquin!
My first attempt was to run the script file (write_sparameters_lumerical.py) in the lumerical folder of the GDSfactory directory. It gave me an error lumapi module not fond. So I used sys.append('api path of lumerical') and added it to the script. The code ran and the FDTD window seems to just open but then closed abruptly with an error.
It seems that the errors are coming from the lumapi.py file(line #242), due to a path/environment issue. Since this is a file in lumerical installation folder, I dont know if I am missing something here. 
[image]
[image]

joamatab, 2023-02-27:
This works great for me
[image]

You can contact Lumerical support 

@mgao-Ansys
@Zeqin-Lu
@121comeon

joamatab, 2023-02-28:
@unrealfancy
@moogie

closed 2023-10-24 by joamatab