gdsfactory/gdsfactory #1363: MacOs and Linux releases failing to build

joamatab, 2023-02-27:
Conda constructor is failing to build MacOs and Linux releases. Only Windows seem to be working. 

I assume it's because Windows machine have more RAM 

https://github.com/gdsfactory/gdsfactory/actions/workflows/release.yml

@jaimergp

jaimergp, 2023-02-27:
Nah I don't think it's memory related. Windows and Linux machines have roughly the same memory, and macOS has 2x-ish. 

It's a conda incompatibility:

```
Traceback (most recent call last):
  File "/usr/share/miniconda/bin/constructor", line 11, in <module>
    sys.exit(main())
  File "/usr/share/miniconda/lib/python3.10/site-packages/constructor/main.py", line 347, in main
    main_build(dir_path, output_dir=out_dir, platform=args.platform,
  File "/usr/share/miniconda/lib/python3.10/site-packages/constructor/main.py", line 172, in main_build
    create(info, verbose=verbose)
  File "/usr/share/miniconda/lib/python3.10/site-packages/constructor/shar.py", line 96, in create
    preconda_write_files(info, tmp_dir)
  File "/usr/share/miniconda/lib/python3.10/site-packages/constructor/preconda.py", line 142, in write_files
    write_index_cache(info, dst_dir, all_dists)
  File "/usr/share/miniconda/lib/python3.10/site-packages/constructor/preconda.py", line 84, in write_index_cache
    write_repodata(cache_dir, url, repodata, used_packages, info)
  File "/usr/share/miniconda/lib/python3.10/site-packages/constructor/conda_interface.py", line 146, in write_repodata
    url = used_repodata.pop('_url').rstrip("/")
KeyError: '_url'
```

Can you pin `conda` to a previous version `<23.1`? [See here](https://github.com/conda/constructor/blob/19350a31ff980b619f3d5dbfeec2b4683747f1f3/conda.recipe/meta.yaml#LL28C7-L28C26).

joamatab, 2023-02-28:
Hi Jaime, I pinned it and still fails to build for Linux and Mac OS

jaimergp, 2023-02-28:
Where did you pin it? Do you have CI logs?

joamatab, 2023-02-28:
https://github.com/gdsfactory/gdsfactory/blob/main/conda/construct.yaml

see logs here

https://github.com/gdsfactory/gdsfactory/actions/runs/4286412604/jobs/7465948354

joamatab, 2023-03-17:
Amazing! thank you @SkandanC 

#1440 

Releases are building now :tada: 

https://github.com/gdsfactory/gdsfactory/releases

joamatab, 2023-04-01:
Hi Jaime, Skandan

after the PR constractor is failing to build relases for all OS
https://github.com/gdsfactory/gdsfactory/actions/runs/4584080213

jaimergp, 2023-04-03:
A new conda-libmamba-solver version was just released, but I think we might need a new `constructor` release too to avoid those conflicts. I'll keep an eye.

joamatab, 2023-04-21:
How about building a mamba version that includes all gdsfactory packages, to replace constructor, similar to mamba?

https://github.com/mamba-org/mamba

joamatab, 2023-06-08:
Floris, i think the SAX link for windows does not seem to work anymore

https://github.com/gdsfactory/gdsfactory/actions/runs/5208126572/jobs/9396371280

joamatab, 2023-09-13:
we recommend installing anaconda

closed 2023-09-13 by joamatab