gdsfactory/gdsfactory #1378: dxf export

joamatab, 2023-03-02:
At the moment we support exporting to STL for 3D files

How can we add the option for DXF export for 3D?
https://pypi.org/project/ezdxf/

HelgeGehring, 2023-03-02:
DXF is not a 3D-format, i.e. it's very similar to GDS

We would need a list where each cell is only once in, then we can probably save them one-by-one as dxf. Do we have that already?
We also need to define the names of the layers, in dxf it's not layer/datatype, but just a string. I've done it like 
```'{:d}-{:d}'.format(layer, datatype)```

Do we use something else than magnification / angle / origin for the references? (x-reflection is not supported in dxf)

HelgeGehring, 2023-03-02:
Besides that: Klayout can directly do it, maybe that's just the solution?

tvt173, 2023-03-02:
an issue i've seen when converting GDS to DXF is that things like circles can blow up with too many points... I know that gdstk can auto-detect and convert circles when writing to OASIS. i wonder if going to OASIS first, then to DXF would be beneficial...