gdsfactory/gdsfactory #1390: Code running for docs is broken

SkandanC, 2023-03-04:
**Describe the bug**
Binder and Colab runs for docs is broken.

**To Reproduce**
Clicking on the Binder/Colab button. It happens because all the documetnation is py files and there are no ipynb files.

**Expected behavior**
It should open Binder/Colab and run the ipynb.

joamatab, 2023-03-04:
Thank you for finding this Skandan!

How can we allow colab to use python jupytext notebooks?

@proppy

closed 2023-03-05 by joamatab

aisichenko, 2023-08-22:
Agreed, [I cannot run the Binder notebooks](https://github.com/gdsfactory/gdsfactory/issues/1390). 

My solution was to create a ipynb in a Google Drive folder which converts the .py files from gdsfactory/docs/notebooks to ipynb:

`!pip install jupytext`
`!jupytext --to ipynb *.py`

For `gdsfactory==7.3.2` I also needed to set the plotting config, because the default `c.plot()` has an error that kweb is not installed. I solved this by running:

`gf.CONF.display_type = 'klayout'`

So, if we can add a cell to all of these .py notebooks that contains these lines then it should run:

```
!pip install gdsfactory
!pip install klayout
import gdsfactory as gf
gf.CONF.display_type = 'klayout'
```

PR #1394 (2023-03-05): remove type annotations for size and live code buttons
fixes #1390 

removes buttons