gdsfactory/gdsfactory #1401: Unnamed cells when extracting layers

simbilod, 2023-03-07:
How do we suppress the `UncachedComponentWarning: Component 'Unnamed_165f14c1' was NOT properly locked. You need to write it into a function that has the @cell decorator.
  warnings.warn(message, UncachedComponentWarning)` when placing a reference to a copied or extracted component?

This happens in the docs as well:

https://gdsfactory.github.io/gdsfactory/notebooks/04_components_geometry.html?highlight=extract#copying-and-extracting-geometry

sebastian-goeldi, 2023-03-08:
For purely suppressing, I would suggest to add a dynamic logfilter similar to this one

https://github.com/gdsfactory/kfactory/blob/main/src/kfactory/config.py#L35

This allows you to either push the level, so that wanrings are not printed, or you can set a regex to match those messages and discard them, like this

https://github.com/gdsfactory/kfactory/blob/main/tests/test_routing.py#L66

String can be a full regex of course

tvt173, 2023-03-08:
since they are warnings, and warnings of a particular class, I think the best way may be to use a warning filter: https://docs.python.org/3/library/warnings.html#overriding-the-default-filter

i.e.
```python
import warnings
from gdsfactory.component import UncachedComponentWarning

warnings.filterwarnings("ignore", category=UncachedComponentWarning)
```

sebastian-goeldi, 2023-03-08:
But they aren't standard warnings, they are loguru warnings, those can't be filtered the standard way

sebastian-goeldi, 2023-03-08:
Oh, yeah, I see. These are actual warnings and not loguru warnings. @joamatab shouldn't these be loguru warnings?

tvt173, 2023-03-08:
we used regular warnings instead of loguru warnings because it is more standard and supported in different ways within python. i.e. we made pytest tests which ensure that the warnings are raised in this situation. but we changed the `show_warning` hook within gdsfactory to print them like loguru warnings. we thought it looked nicer and more consistent that way, but maybe it is a bit confusing though?

sebastian-goeldi, 2023-03-08:
One thing to keep in mind then. The default settings for warnings will only raise a warning once (unless you changed that when modifying the hook).

I don't have a strong opinion on which warning to use though.

tvt173, 2023-03-08:
it actually doesn't happen when the cell is created, only when it is _written to GDS_. but maybe we should suppress it when calling `show()`? I think it is appropriate when calling `write_gds()` in other scenarios

we discussed but didn't implement an `'ignore'` option for the `on_uncached_component` [GDS write flag](https://github.com/gdsfactory/gdsfactory/blob/main/gdsfactory/pdk.py#L55). maybe we should implement it, then pass `on_uncached_component="ignore"` when calling `write_gds` within `show()`?

tvt173, 2023-03-08:
> it actually doesn't happen when the cell is created, only when it is _written to GDS_. but maybe we should suppress it when calling `show()`? I think it is appropriate when calling `write_gds()` in [...]

@simbilod @joamatab , please see https://github.com/gdsfactory/gdsfactory/pull/1402, which addresses this.

tvt173, 2023-03-08:
A few things I've come to realize after digging a bit deeper on this...

1. The current implementation will not actually warn or throw an error if the _top level_ is uncached. Is this a bad thing? Maybe not. It's usually not consequential if your top level is uncached and it may be reasonable to do, i.e. if writing a copy of a cell.
2. That considered, it should *almost* be fine not to make this change... we might often expect (i.e. when showing a component in a notebook) for the _top level_ to be uncached, and so if that does not throw a warning, we should not need to forego them. We still expect that _intermediate cells_ should not be uncached in almost any circumstance though. But what screws this up a bit is `show_ports`, which will push the top level one layer down the hierarchy, and thus the warning will be thrown.
3. There are also examples in the notebooks which use exactly the bad practices we are trying to prevent i.e. https://gdsfactory.github.io/gdsfactory/notebooks/04_components_geometry.html?highlight=extract#xor-diff 
[image] 

Now, to the last point, I don't think it's a good idea that we have examples in the notebooks which use bad practices. However, I think it was probably done this way because it felt too verbose within an example to do this properly, declaring functions for A and B, which is something I think users sometimes encounter also. Wouldn't it be nice to still be able to create components inline like this in certain circumstances, but in a safe way? What do you think ? I'm thinking we could do this with a context manager, like below:

```python
import gdsfactory as gf

with gf.create_component("A") as A:
  A.add_ref(gf.components.ellipse(radii=[10, 5], layer=(1, 0)))
  A.add_ref(gf.components.text("A")).move([3, 0])

with gf.create_component("B") as B:
  B.add_ref(gf.components.ellipse(radii=[11, 4], layer=(1, 0))).movex(4)
  B.add_ref(gf.components.text("B")).move([3.2, 0])
X = gf.geometry.xor_diff(A=A, B=B, precision=1e-6)

# Plot the original geometry and the result
# Upper left: A / Upper right: B
# Lower left: A and B / Lower right: A xor B "diff" comparison
D = gf.Component("xor_diff")
D.add_ref(A).move([-15, 25])
D.add_ref(B).move([15, 25])
D.add_ref(A).movex(-15)
D.add_ref(B).movex(-15)
D.add_ref(X).movex(15)
D
```

`create_component()` could roughly look something like this

```python
@contextmanager
def create_component(name):
    # create the component
    c = gf.Component()
    try:
        yield c
    finally:
        # register/cache the component and lock it
        gf.register_component(name, c)
        c.lock()
```

What do you think?

joamatab, 2023-03-08:
I think we should keep the tutorial as simple as possible

How about doing this?

```
import gdsfactory as gf

@gf.cell
def demo_text()
  A = gf.Component()
  A.add_ref(gf.components.ellipse(radii=[10, 5], layer=(1, 0)))
  A.add_ref(gf.components.text("A")).move([3, 0])
  return A

@gf.cell
def demo_xor():
  B = gf.Component()
  B.add_ref(gf.components.ellipse(radii=[11, 4], layer=(1, 0))).movex(4)
  B.add_ref(gf.components.text("B")).move([3.2, 0])
  X = gf.geometry.xor_diff(A=A, B=B, precision=1e-6)
  return B

# Plot the original geometry and the result
# Upper left: A / Upper right: B
# Lower left: A and B / Lower right: A xor B "diff" comparison
D = gf.Component("xor_diff")
D.add_ref(A).move([-15, 25])
D.add_ref(B).move([15, 25])
D.add_ref(A).movex(-15)
D.add_ref(B).movex(-15)
D.add_ref(X).movex(15)
D

```

Also, at the moment we don't lock the components that are imported from gds, which is also showing lots of warnings, what do you think of locking any new component cell created after importing it from GDS?

tvt173, 2023-03-08:
hmm, I think the issue we had with _caching_ imported GDS files was that there could be name collisions, i.e. if you created a cell, then saved it and opened it. if we don't uniquely rename all the cells in the imported GDS, it would overwrite or be replaced by cells already in the cache.

i'm looking back at this, and I see the original example you posted [here ](https://github.com/gdsfactory/gdsfactory/issues/1306) _already did_ throw a warning about **duplicate cell names**, which is ultimately the problem that is causing the subtle error you point out in that example. for the average user, isn't that already enough? i think the new check is nice for those of us who want to be very strict with our methodologies to prevent _future_ errors, but ultimately the error that it is trying to prevent is duplicate cell names within one GDS library, which we are already warning about. maybe it is too strict and redundant to throw an additional warning for this by default. what if we just set the default value for `on_uncached_component` to ignore?

tvt173, 2023-03-08:
the other thing you may want to consider is throwing an **error** rather than a warning when writing a GDS with **duplicate cell names** by default. it's hard for me to think of scenarios where it is acceptable for this to happen

sebastian-goeldi, 2023-03-08:
I agree with @tvt173 . A gds with two identical cells in it will not load in klayout 0.27+ and will throw an error that for most people might be rather cryptic

tvt173, 2023-03-09:
your other option, for PDKs build from many imported GDS files, would be to do a one-time merge of all the input GDS files into one database, checking that all the common-named cells are indeed the same. then import and cache all the cells (even subcells) to prevent errors from happening later on. since this is really a problem i think with this one isolated use case, something like this might make sense

closed 2023-10-23 by joamatab