gdsfactory/gdsfactory #1426: pip install gdsfactory; fails on OSX M1

lukasc-ubc, 2023-03-15:
@joamatab or @thomaslima

I tried installing gdsfactory on Mac OSX, M1, Monterey, using:
`pip3.9 install gdsfactory`
and
`pip3.10 install gdsfactory`

Both give me an error:
```
      clang: error: no such file or directory: 'build/temp.macosx-12-arm64-cpython-310/src/tl/tl/tlPixelBuffer.o'
      error: command '/usr/bin/clang++' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for klayout
Successfully built gdstk
Failed to build klayout
ERROR: Could not build wheels for klayout, which is required to install pyproject.toml-based projects
```

Looking above for the tlPixelBuffer file, has the following error:
 ```
     clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk -dynamiclib -DHAVE_PNG=1 -DHAVE_CURL=1 -DHAVE_EXPAT=1 -DKLAYOUT_MAJOR_VERSION=0 -DKLAYOUT_MINOR_VERSION=28 -DGSI_ALIAS_INSPECT=1 -DMAKE_TL_LIBRARY=1 -I/opt/homebrew/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/include/python3.9 -c src/tl/tl/tlStaticObjects.cc -o build/temp.macosx-12-arm64-cpython-39/src/tl/tl/tlStaticObjects.o -Wno-strict-aliasing -std=c++11
      clang: warning: argument unused during compilation: '-dynamiclib' [-Wunused-command-line-argument]
      src/tl/tl/tlPixelBuffer.cc:28:12: fatal error: 'png.h' file not found
      #  include <png.h>
                 ^~~~~~~
      1 error generated.
      Building build/temp.macosx-12-arm64-cpython-39/src/tl/tl/tlPixelBuffer.o has failed. Trying again.
      Building build/temp.macosx-12-arm64-cpython-39/src/tl/tl/tlPixelBuffer.o
      clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk -dynamiclib -DHAVE_PNG=1 -DHAVE_CURL=1 -DHAVE_EXPAT=1 -DKLAYOUT_MAJOR_VERSION=0 -DKLAYOUT_MINOR_VERSION=28 -DGSI_ALIAS_INSPECT=1 -DMAKE_TL_LIBRARY=1 -I/opt/homebrew/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/include/python3.9 -c src/tl/tl/tlPixelBuffer.cc -o build/temp.macosx-12-arm64-cpython-39/src/tl/tl/tlPixelBuffer.o -Wno-strict-aliasing -std=c++11
      clang: warning: argument unused during compilation: '-dynamiclib' [-Wunused-command-line-argument]
      Building build/temp.macosx-12-arm64-cpython-39/src/tl/tl/tlScriptError.o

```

png.h comes from libpng, which I have installed:
```
% brew install libpng
==> Downloading https://formulae.brew.sh/api/formula.jws.json
######################################################################## 100.0%
==> Downloading https://formulae.brew.sh/api/cask.jws.json
######################################################################## 100.0%
Warning: libpng 1.6.39 is already installed and up-to-date.
To reinstall 1.6.39, run:
  brew reinstall libpng
```

Perhaps there is a way to disable including klayout?  

I am actually trying to install gdsfactory within KLayout, rather than KLayout within gdsfactory...

thank you
Lukas

lukasc-ubc, 2023-03-15:
I was able to get gdsfactory working within KLayout's IDE, using a compiled LW KLayout:

- required a fork of both KLayout and GDSfactory
  - https://github.com/lukasc-ubc/klayout
  - https://github.com/lukasc-ubc/gdsfactory

- my notes: https://github.com/lukasc-ubc/klayout/blob/master/macbuild/lukasc_notes.txt
  - in KLayout, "Add Location" to the GitHub folder
  - add gdsfactory to the system path, then import

Next, 
1) would be great to do this in the HW version, on Windows, Linux and OSX, with all the Python packages pre-loaded (it is a really long list)
2) have GDSfactory PCells show up in a KLayout library, so we can drag and drop.
3) have GDSfactory output directly into the KLayout database, rather than saving to disk.

Exciting!

@joamatab @mustafacc @thomaslima

p.s.  "pip install gdsfactory" still doesn't work for me...

[image]

joamatab, 2023-03-15:
Hi Lukas,

for gdsfactory klayout is an optional dependency, we can have it as `pip install gdsfactory[full]`

let me make a new release and you can try again

joamatab, 2023-03-15:
just released `6.60.0` which does not require klayout, let me know if `pip install gdsfactory` works now on m1

lukasc-ubc, 2023-03-15:
works on M1, thank you!

closed 2023-03-15 by joamatab

lukasc-ubc, 2023-03-15:
Hi Joaquin,

Just so you know, I tried:

`pip3.9 install 'gdsfactory[full]'`

using version 6.60.0, and I get the same error with KLayout.  So it isn't possible today to get GDSfactory with KLayout as a viewer working on OSX M1:

```
      clang: error: no such file or directory: 'build/temp.macosx-12-arm64-cpython-39/src/tl/tl/tlPixelBuffer.o'
      error: command '/usr/bin/clang++' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for klayout
```

joamatab, 2023-03-16:
Hi Lucas, that will get fixed next week with the 0.28.6 klayout release

https://github.com/KLayout/klayout/issues/1314

PR #1429 (2023-03-15): klayout is an optional dependency
should fix #1426