gdsfactory/gdsfactory #1428: Surface gdstk fillet tools

95-martin-orion, 2023-03-15:
**Is your feature request related to a problem? Please describe.**
The `gdstk.Polygon` object provides a [`fillet` method](https://heitzmann.github.io/gdstk/geometry/gdstk.Polygon.html#gdstk.Polygon.fillet) to fillet corners of the polygon. Currently, the only way I'm aware of to use this feature in gdsfactory is to instantiate a `gdstk.Polygon`, call `fillet` on it, then pass the resulting points to `gf.Component.add_polygon`.

**Describe the solution you'd like**
Either of the following would work:
- A `@fillet` decorator which triggers filleting on all geometry created inside of it
- Fillet parameters (similar to those used in `gdstk.Polygon.fillet`) on basic geometry methods (bbox, add_polygon, etc.)

joamatab, 2023-03-21:
Hi Orion,

how about adding `Component.fillet()` method?

95-martin-orion, 2023-03-21:
My suggestions above were based on my perception that component geometry seems "fixed" once it's created. If that's not a strict requirement, `Component.fillet()` would also work.

joamatab, 2023-03-21:
it's only fixed (locked) after returned by a function that has the cell decorator

How about return a copy of the component and do the fillet on it?

95-martin-orion, 2023-03-21:
> How about return a copy of the component and do the fillet on it?

This would also work, though putting `fillet` at the Component level instead of geometry level may require separating a design into more components than would otherwise be necessary. For example, if a component contains one shape which requires filleting and one shape which does not, the shape to be filleted would need to be converted into a subcomponent if `Component.fillet()` or similar is used.

I'm unfortunately not a hardware specialist, so I can't really speak to how often a design could simply call `fillet` on every polygon it contains vs. having to manually identify and fillet polygons one-by-one.

closed 2023-03-23 by joamatab