gdsfactory/gdsfactory #1432: keep the polarity on spirals with doped slabs on the bends

joamatab, 2023-03-16:
How can we keep the polarity on spirals with doped slabs on the bends?

```
import gdsfactory as gf

if __name__ == "__main__":
    c = gf.components.spiral_inner_io(
        cross_section=gf.cross_section.pin, waveguide_spacing=25, radius=30
    )
    c.show()

```

[image]

@SkandanC

simbilod, 2023-03-16:
[image]

```
if __name__ == "__main__":

    c = gf.components.spiral_inner_io(
        cross_section=gf.cross_section.pin, 
        waveguide_spacing=25, 
        radius=30, 
        cross_section_bend=gf.partial(gf.cross_section.pin, mirror=True),
        asymmetric_cross_section=True
    )
    c.show()

```

See #1049 and #1433

simbilod, 2023-03-16:
It would be nice to take another look at the general solution at some point https://github.com/gdsfactory/gdsfactory/issues/1049#issuecomment-1366908027

closed 2023-03-16 by joamatab