gdsfactory/gdsfactory #1465: faster dummy fill tiling

joamatab, 2023-03-23:
gdsfactory has `gf.fill` that allows you to [dummy fill](https://gdsfactory.github.io/gdsfactory/notebooks/04_components_geometry.html#dummy-fill-tiling) but for big structures this can be a bit slow

how  could we leverage the fill capabilities from klayout?
https://www.klayout.de/doc-qt5/manual/tiling.html#k_2

One issue is that the fill tool is not exposed in the python API
How about using some subprocess calls to handle this?

Similar to [lymask](https://github.com/atait/lymask/blob/master/lymask/dataprep_steps.py)

@sebastian-goeldi 
@klayoutmatthias
@SkandanC

sebastian-goeldi, 2023-03-23:
Fill API is not exposed in python? Could you elaborate @joamatab ? There is https://www.klayout.de/doc-qt5/code/class_Cell.html#method69 and https://www.klayout.de/doc-qt5/code/class_Region.html#method71 and their multi variants that are fully available in python. The thing you can't use (afaik) is the tiling processor to speed it up with multithreading. I am not sure that is desirable in this context

klayoutmatthias, 2023-03-23:
Yes, the fill tool is exposed to Python, but in a more atomic fashion.

The fill pattern needs to be supplied through a fill cell which is arrayed using AREFs (to some extend). Enhanced and multi-pass fill can be implemented with "fill_region_multi" and by using the remaining (non-filled) parts and running over them with a second (typically smaller) fill pattern.

I am not sure if multi-threading is fully supported. Basically it is, but the fill pattern insertion needs to employ a Mutex which I suspect is not present yet. I will check this.

Matthias

sebastian-goeldi, 2023-03-23:
@joamatab 

For code examples, see https://www.klayout.de/doc-qt5/code/class_Cell.html#m_fill_region ,

translated to python:

```python

r = ...        # region to fill
c = ...        # cell in which to produce the fill cells
fc_index = ... # fill cell index
fc_box = ...   # fill cell footprint

fill_margin = pya.Point(0, 0)   # x/y distance between tile cells with different origin

# Iteration: fill a region and fill the remaining parts as long as there is anything left.
# Polygons not worth being considered further are dropped (last argument is nil).
while not r.empty():
  c.fill_region(r, fc_index, fc_box, nil, r, fill_margin, nil)
end
```

You can also try to use the direct multifill with

```python

r = ...        # region to fill
c = ...        # cell in which to produce the fill cells
fc_index = ... # fill cell index
fc_box = ...   # fill cell footprint
fc_row_vec = pya.Vector(fc_box.width(), 0)
fc_col_vec = pya.Vector(0, fc_box.height())

c.fill_region_multi (r, fc_index,fc_box,fc_row_vec, fc_col_vec)
```

joamatab, 2023-03-23:
I tried using kfactory for that but did not get the fill working

let me know if im missing something

```
from kfactory import KCell, kdb
import kfactory as kf
import klayout.db as kdb

@kf.autocell
def fill(cell_to_fill: KCell, fill_cell: KCell) -> KCell:
    """Return a new Cell that has fill on it."""
    c = KCell("bend_with_fill")
    fc_index = fill_cell.cell_index()  # fill cell index

    c.fill_region(r, fc_index, fc_box, None, r, fill_margin, None)
    return c

if __name__ == "__main__":
    from kfactory.pcells import waveguide, euler

    c = KCell("bend_with_fill")
    layer = (1, 0)
    layer = kf.klib.layer(*layer)
    c << euler.bend_euler(width=0.5, radius=5, layer=layer)
    w = waveguide.waveguide(width=3, length=3, layer=layer)
    fc_index = w.cell_index()  # fill cell index
    fc_box = w.bbox()  # fill cell footprint
    fill_margin = kdb.Point(
        0, 0
    )  # x/y distance between tile cells with different origin

    # Iteration: fill a region and fill the remaining parts as long as there is anything left.
    # Polygons not worth being considered further are dropped (last argument is nil).
    reg = kdb.Region()  # region to fill

    c.fill_region(reg, fc_index, fc_box, None, reg, fill_margin, None)
    c.show()

```

sebastian-goeldi, 2023-03-23:
Yeah,don't tell KLayout to fill an empty region ;)

```python
from kfactory import KCell, kdb
import kfactory as kf
import klayout.db as kdb

@kf.autocell
def fill(cell_to_fill: KCell, fill_cell: KCell) -> KCell:
    """Return a new Cell that has fill on it."""
    c = KCell("bend_with_fill")
    fc_index = fill_cell.cell_index()  # fill cell index

    c.fill_region(r, fc_index, fc_box, None, r, fill_margin, None)
    return c

if __name__ == "__main__":
    from kfactory.pcells import waveguide, euler

    c = KCell("bend_with_fill")
    layer = (1, 0)
    layer = kf.klib.layer(*layer)
    c << euler.bend_euler(width=0.5, radius=5, layer=layer)
    w = waveguide.waveguide(width=3, length=3, layer=layer)
    fc_index = w.cell_index()  # fill cell index
    fc_box = w.bbox()  # fill cell footprint
    fill_margin = kdb.Point(
        0, 0
    )  # x/y distance between tile cells with different origin

    # Iteration: fill a region and fill the remaining parts as long as there is anything left.
    # Polygons not worth being considered further are dropped (last argument is nil).
    reg = kdb.Region()  # region to fill

    c.fill_region(reg, fc_index, fc_box, None, reg, fill_margin, None)
    c.show()
```

btw, you can tell github to format something as python code with \`\`\`python instead of only the backticks on the first line of the code block

Also if you don't iterate on  this, you will not fill all available slots, I suggest to use the multi fill, that one works well

joamatab, 2023-03-24:
This is great, thank you Sebastian

also this looks great, thank you for the tip

```python
print('thank you')
```

joamatab, 2023-03-25:
Sebastian, How can I reproduce the plot above with the code that you posted?

This is what im trying

```python
    c = KCell("bend_with_fill")
    layer = (1, 0)
    layer = kf.klib.layer(*layer)
    c << euler.bend_euler(width=0.5, radius=5, layer=layer)
    w = waveguide.waveguide(width=3, length=3, layer=layer)
    fc_index = w.cell_index()  # fill cell index

    # fc_box = w.bbox()  # fill cell footprint
    fc_box = w.bbox().enlarged(100, 100)
    fill_margin = kdb.Point(
        0, 0
    )  # x/y distance between tile cells with different origin

    # Iteration: fill a region and fill the remaining parts as long as there is anything left.
    # Polygons not worth being considered further are dropped (last argument is nil).
    region = kdb.Region()  # region to fill
    region.insert(c.begin_shapes_rec(layer))
    c.fill_region(region, fc_index, fc_box, None, region, fill_margin, None)
    c.write('fill.gds')

```

And this is what I get

[image]

sebastian-goeldi, 2023-03-25:
Sorry, I think I copied the wrong code

```python
import kfactory as kf

c = kf.KCell("bend_with_fill")
layer = (1, 0)
layer = kf.klib.layer(*layer)
c << kf.pcells.euler.bend_euler(width=0.5, radius=5, layer=layer)
w = kf.pcells.waveguide.waveguide(width=0.3, length=0.3, layer=layer) # note, this is a smaller cell as 3um is gigantic compared to the bend
fc_index = w.cell_index()  # fill cell index

# fc_box = w.bbox()  # fill cell footprint
fc_box = w.bbox().enlarged(100, 100)
fill_margin = kf.kdb.Point(
    0, 0
)  # x/y distance between tile cells with different origin

# Iteration: fill a region and fill the remaining parts as long as there is anything left.
# Polygons not worth being considered further are dropped (last argument is nil).
region = kf.kdb.Region(c.begin_shapes_rec(layer))  # region to fill
region = kf.kdb.Region(c.bbox()) - region
c.fill_region(region, fc_index, fc_box, None, region, fill_margin, None)
# c.write('fill.gds')
c.show()
```

[image]

simbilod, 2023-03-25:
see #1484 , thank you Troy and Sebastian. If you think (or when you realize) it can be done better let us know :)

joamatab, 2023-03-26:
How could we pass a specific fill cell, as well as layers to avoid the fill?

similar to the functions `gf.fill`

SkandanC, 2023-03-26:
What are we trying to do with the fill function here? Is it returning a KCell? If so, it should be noted that c.fill_region returns None.

sebastian-goeldi, 2023-03-26:
Specific cell: `kf.klib[cellname]` for any cell that is in the layout. Layers to avoid: just chain them into the region to substract (or chain the subtractions)

sebastian-goeldi, 2023-03-26:
@joamatab how strongly is it necessary to back support python >=3.8? kfactory uses the match case even for the base import (for yaml <-> kcell), and atm I don't have the time to backport those cases (and the readability will suffer greatly probably).

joamatab, 2023-03-26:
it's okay to drop 3.8 and 3.9 support, we already have warnings in place

https://github.com/gdsfactory/gdsfactory/blob/main/gdsfactory/dataprep.py

simbilod, 2023-03-26:
I've been doing everything with layers, like adding a bbox around my gdsfactory component on a dummy "floorplan" layer and then defining a region to fill that is that layer - layers I don't want filled

closed 2023-04-26 by joamatab