gdsfactory/gdsfactory #1466: gerber files

joamatab, 2023-03-23:
How can we create files with gdsfactory?

GDS represents circles/arcs as polygons, but Gerber stores center + radius

@95-martin-orion
@nikosavola 
@elamdf

tvt173, 2023-03-23:
gdstk supposedly has the ability to convert circle-like polygons to true circles when writing to OASIS. so maybe something can be done by first writing to OASIS and using the resulting shapes, but i'm not quite sure how to get access to them...

95-martin-orion, 2023-03-27:
There's a further complication here in that circles are not the only ideal curve Gerber supports. The [Gerber spec](https://www.ucamco.com/en/guest/downloads/gerber-format) also includes "circular plot" mode in section 4.7.2, which allows the creation of regions whose boundary is a mix of straight lines and curves - for example, a filleted rectangle or a curved wire segment.

These shapes won't be captured by gdstk circle detection, which suggests one of the following solutions:

1. More complex detection and conversion tools in gdstk
    - Unless the gdstk devs already plan to fully support Gerber, this seems like a difficult request.
1. Gerber conversion at the gdsfactory level
    - This is similarly complicated, but it's at least in the gdsfactory domain.
1. As (2), but restricted to a set of "Gerber-friendly" components
    - Much easier to implement, but likely won't support all use cases. Could allow more exact conversions.

Given that Gerber doesn't show up in the gdstk issue list, (3) seems like the clearest path forward to me. It would unblock the Gerber use case while potentially building towards a more general (2)-style solution.

joamatab, 2023-03-27:
We have been also looking at switching from gdstk to klayout backend with kfactory

i noticed that Klayout has gerber support

@klayoutmatthias

klayoutmatthias, 2023-03-27:
It does, but it will convert everything to polygons. Gerber offers things like programmable apertures which are beyond what native shapes can support.

Matthias

joamatab, 2023-09-16:
Only one PR was made, we can re-open and are looking forward for contributions on this

95-martin-orion, 2023-11-15:
Brushing off the dust - this is still a feature my team is interested in.

daquintero, 2024-07-10:
Just happened to be around here and just as an idea, there could be a scope of a kicad gplugins in this context, integrating:

* https://github.com/yaqwsx/KiKit
* eg. rf applications like https://forum.kicad.info/t/dxf-import-issue/20559/2

Maybe there's a way to  automate compoent to gds <-> kicad accordingly. Probably doesn't solve the gds polygon transformations though but at least can be a kicad bug than a kfactory/gdsfactory issue.