gdsfactory/gdsfactory #1472: improve dataprep

joamatab, 2023-03-24:
The idea is to use klayout engine to do mask preparation steps (booleans, sizing ...)

some other repos with similar concepts:

 - [BPG](https://github.com/BerkeleyPhotonicsGenerator/BPG/blob/master/BPG/examples/tech/BPG_tech_files/dataprep_routine.yaml)
  - [lymask](https://github.com/atait/lymask/blob/master/lymask/dataprep_steps.py)

Things to do:

- [ ] manhattanize

Maybe:

- [ ] expand with fill

Based on
https://github.com/gdsfactory/gdsfactory/pull/1468

@tvt173
@mdecea
@dkramnik
@simbilod
@nikosavola
@sebastian-goeldi
@elamdf
@atait
@pvnbhargava

simbilod, 2023-03-25:
see #1484 for fill draft

joamatab, 2023-03-26:
looks great! thank you Simon!

sebastian-goeldi, 2023-03-28:
> we would either need to support multiprocessing backend by implementing serialization of Region objects or get klayout to [release the GIL](https://www.geeksforgeeks.org/releasing-gil-and-mixing-threads-from-c-and-python/) during these operations (the latter option is probably [preferable](https://distributed.dask.org/en/stable/worker.html#thread-pool) if possible)

This is not possible for the creation of the regions as for how gdsfactory currently does it. It reads it with a recursive shape iterator which by definition is not thread safe. Merges etc could be made thread safe. In my experience it is usually not the construction of the regions that takes the most time but rather the filling, so I think it is at the moment not the right path to go. If you want to use threads for the filling, we can look into using the TilingProcessor, which Matthias confirmed to be available in python and that the necessary mutex is in place.

> In either case, I would argue that if we support dask, our Region class should not inherit from Klayout's Region

Fully agree on this. There are also quite a few things that could use some serious cleanup. A list that is probably not exhaustive:
* hardcoding the DBU -> um is just an all around bad idea. It will seriously mess up your layout if your DBU is not 1nm
* sizing a region is not guaranteed to give you non-self-intersecting polygons. E.g. in a bend with a high point density the sizing operation will mess up the sizing of the inner bend radius. This can be avoided with minkowski sums, but to get a good representation the sizing operation it is very expensive to do so. If you are purely using it for the fill, it should be fine for not gigantic sizings (sizing a 2um radius bend by 2um would probably not be fine).
* there is a lot of region duplication going on. E.g. a sizing operation doesn't require a new region if you discard the old region anyway. In this case you are just waiting for memory to allocate you new heap, which takes in the order of 10-100ns even for small memory areas. For large memory areas you will be waiting the majority of time probably.

tvt173, 2023-03-29:
i'm not experienced enough with klayout's derivations/drc to have a good sense of what to expect as far as runtimes (though I do recall seeing that sizing could also be rather expensive for large growth sizes... not sure how this might change with minkowski sums...). i'd be interested to try this out on a bigger example. hopefully i can carve out some time to do this in the next few weeks. but if we go down a similar path for DRC, I think we can start to expect long runtimes which warrant some sort of parallelization scheme, and task-based parallelization probably makes sense for that use case (a DRC deck with hundreds of rules which can be executed independently...). but for DRC decks, a declarative definition (i.e. table or yaml file) also probably makes more sense.

tvt173, 2023-03-29:
and so to finish that thought, maybe it makes sense to remove dask support now, and implement a parallelism scheme for DRC deck execution if/when we add it. that is a much clearer use case that would clearly benefit from it, and we could tailor it to be efficient for that use case. it sounds like for fill also, there are other parallelization schemes already in klayout which we could use which would also be more efficient for that use case

sebastian-goeldi, 2023-03-29:
Regarding sizing there are some tricks you can do. You can use a deep shap storage to do it hierarchical. That should speed it up considerably (in most cases). And also you can use the tiling processor to parallelize. In some cases it can cause problems in the overlapping region between the tiles though. It also helps if independent structures are not in the same region etc.

joamatab, 2023-08-11:
moving it here

https://github.com/gdsfactory/gplugins/issues/32

closed 2023-08-11 by joamatab