gdsfactory/gdsfactory #1497: Isolate klive from gdsfactory?

SkandanC, 2023-03-28:
**Is your feature request related to a problem? Please describe.**
Currently gdsfactory suggests installing klive either through the cli, or through the package manager (which is v5.47.0  several months old). KFactory suggests installing klive through the KLayout salt package manager. Despite being the same scripts afaik, the gdsfactory klive doesn't work with kfactory. However, klive from the package manager seems to work for both.

**Describe the solution you'd like**
Isolate klive from gf, and have it maintained through the klive repo.

sebastian-goeldi, 2023-03-28:
Did you try whether it works with gdsfactory? Also, might be a good idea to patch gdsfactory to do a stupid easy "handshake" like kfactory does.  klive of gdsfactory does not work with kfactory because it expects that handshake. It should still work, but might clean up the gds too fast

joamatab, 2023-03-29:
yes, I tried the klive in the SALT klayout package and it works with gdsfactory

i didn't like that it does not keep the position when loading a new GDS so i decided to keep using the one in  gdsfactory

once that is fixed on the klayout package im happy to switch and keep using/maintaing only one

sebastian-goeldi, 2023-03-29:
Could you explain the not keeping the position? It should keep the position and even cell selection and layer selection as long as the file names are the same. If the file names are not the same, it doesn't make sense to keep the position, does it?

joamatab, 2023-03-29:
i prefer always keeping position and layer selection even if file names are not the same

joamatab, 2023-05-01:
it would be great to make the klive with the gdsfactory klive compatibles

PR #1612 (2023-05-04): Remove klive
remove klive from gdsfactory

from now on we'll recommend installing klive from klayout package manager

fixes #1497 

@sebastian-goeldi 
@SkandanC
closed 2023-05-04 by joamatab