gdsfactory/gdsfactory #1504: Improve maskprep documentation

joamatab, 2023-03-30:
how can we clean min gaps and acute angles on slab/cladding layers?

I've been trying doing it at the cell level but is hard to get all the cases right, specially when routing between cells
using bbox layers instead of cladding_layers can help but not always

https://gdsfactory.github.io/gdsfactory/notebooks/dataprep.html#remove-acute-angles

I think it would be better doing it at the top level in most cases,
However with klayout python API is a bit slow because single core, What do you think of using a ruby DRC script instead and running multicore

the other option is doing this hierarchically
https://github.com/gdsfactory/kfactory/blob/main/src/kfactory/utils/violations.py#L4-L20

joamatab, 2023-04-02:
yes, in both cases the script did not finish in 10-20min on a 5x5mm layout

how can we speed it up?

sebastian-goeldi, 2023-04-02:
That is very slow. It should finish in the order of 1min maximum

joamatab, 2023-04-03:
Actually after 1.5hours the github action got cancelled :)

sebastian-goeldi, 2023-04-03:
That probably mean an iterator wasn't advamced when it should.

joamatab, 2023-08-11:
moving it to this issue

https://github.com/gdsfactory/gplugins/issues/32

closed 2023-08-11 by joamatab