gdsfactory/gdsfactory #1509: Manhatanize a layer?

joamatab, 2023-03-31:
How can we manhatanize a layer?

See for example how BPG does it

[BPG](https://github.com/BerkeleyPhotonicsGenerator/BPG/tree/master/BPG/compiler)

@SkandanC 
@lucas-flexcompute 
@mdecea

joamatab, 2023-04-02:
[image]

sebastian-goeldi, 2023-04-02:
I have C++ code that can do it, but it needs a shared memory interface, not very suitable. Is there even still a big need for manhattanization? Most photonic pdks support freeform nowadays, don't they?

joamatab, 2023-04-02:
How would you fix acute angles on slabs/claddings/enclosures?

if min_gap is 0.5 we can manhattanize with 1um gap and then we never have gaps below that

[image]

sebastian-goeldi, 2023-04-02:
The klayout/kfactory script should be able to very easily fix it. Can you show me a code piece where it doesn't? I have fixed tons of these kinds of errors withit before, and it shouldn't be slow, especially if you call it on specific cells

closed 2023-04-26 by joamatab