gdsfactory/gdsfactory #1512: register @cell functions

joamatab, 2023-03-31:
How can we register `@cell` functions, to make sure there are not two cells

they can cause name conflicts

sebastian-goeldi, 2023-04-02:
Easiest way is probably to have a global cache (easiest version is a dict) and then check whether the local cache (`@cell`) and the global cache produce the same cell for a name and if not throw an error.

Vivswan, 2023-05-25:
I encountered this issue as well because of how the cache uses `get_component_spec`, which in turn leads to the fact that `name` is being used as `id`.

There may be situations where users want the same name for different components. (like multiple different types alignment marks, just named `alignment marker`  )
Therefore, adding a new `id` parameter makes more sense according to me.

This new "id" has to be based on the content of the cell, but implementing this may be quite complicated. The quick and dirty approach would be:

- If the cell is empty, then use `sha-256(self.uid)`.
- If the cell does not have any polygon or additional features itself, then use `sha-256(sorted(children_ids)).`
- If the cell has its own polygon or additional features, then use `sha-256(sorted(children_ids + [self.uid]))`.

tvt173, 2023-05-25:
yes, but there is some method to all this madness... let's take a minute to reflect on the hopes and expectations of the average user:

## Hope No. 1

> My component's name stays the same between sessions.

Not sure exactly what you mean by your first bullet point, but usually a uid is generated randomly, which would give the component a new name each time the layout is generated (even though it remains exactly the same). This quickly becomes annoying and difficult to track.

## Hope No. 2

> My component's name doesn't change from a minor tweak to the internal code/geometry.

If it did, again, it becomes difficult and annoying to track. Think about the task of implementing regression tests in this scenario... So typically we wish the name of our component to remain constant for a given _configuration_ = name + parameters. Sometimes we change the content of our cell so much that we may wish to name it differently, but that is left up to the discretion of the PDK designer/organization to do explicitly, i.e. between revisions of their PDK.

So the idea of naming based on a hash of the geometries is something we also considered in the past, but this is where it breaks.

-------------

I think the main concern here should be implementing a mechanism to warn or throw an error if a component is defined a second time in a PDK without it being done (or excused) very explicitly. There can be legitimate reasons to do this... i.e. if I want to base my PDK off the generic PDK but redefine the `straight` component to something slightly different. Right now you are allowed to do this. But it might be good to enforce that when you are doing it, you are doing it very explicitly, so that it is not done on accident (i.e. accidentally adding a component with the same name to a different module of the same PDK).

closed 2024-01-01 by github-actions[bot]