gdsfactory/gdsfactory #1534: Klayout fill

joamatab, 2023-04-08:
How can we leverage klayout fill to fill big masks?

This is the script I've been trying to use

```python

"""Dummy fill to keep density constant using klayout."""
from __future__ import annotations

from typing import Optional, Tuple

import gdsfactory as gf
from gdsfactory.typings import PathType, LayerSpecs, Layer

import kfactory as kf
import klayout.db as kdb

def fill(
    gdspath,
    layer_to_fill: Layer,
    layer_to_fill_margin: float = 0,
    layers_to_avoid: LayerSpecs = None,
    layers_to_avoid_margin: float = 0,
    cell_name: Optional[str] = None,
    fill_cell_name: Optional[str] = None,
    fill_layers: LayerSpecs = None,
    fill_size: Tuple[float, float] = (10, 10),
    fill_spacing: Tuple[float, float] = (20, 20),
    fill_name: str = "fill",
    gdspath_out: Optional[PathType] = None,
) -> None:
    """Write gds file with fill.

    Args:
        gdspath: GDS input.
        gdspath_out: Optional GDS output. Defaults to input.
        cell_name: Optional cell to fill. Defaults to top cell.
        fill_cell_name: Optional cell name to use as fill.
        fill_layers:
        fill_size:
        fill_spacing:
        fill_name: name of the cell containing all fill cells.
    """

    lib = kf.kcell.KLib()
    lib.read(filename=str(gdspath))
    cell = lib[cell_name or 0]

    if fill_cell_name:
        fill_cell = lib[fill_cell_name]

    else:
        fill_cell = kf.KCell("fill_cell")
        for layer in fill_layers:
            layer = gf.get_layer(layer)
            layer = kf.klib.layer(*layer)
            fill_cell << kf.pcells.waveguide.waveguide(
                width=fill_size[0], length=fill_size[1], layer=layer
            )

    fill_cell_index = fill_cell.cell_index()  # fill cell index
    fill_cell_box = fill_cell.bbox().enlarged(
        fill_spacing[0] / 2 * 1e3, fill_spacing[1] / 2 * 1e3
    )
    fill_margin = kf.kdb.Point(0, 0)

    region = kdb.Region()
    layer = gf.get_layer(layer_to_fill)
    layer = kf.klib.layer(*layer)
    region.insert(cell.begin_shapes_rec(layer))
    region.merge()
    # region.size(layer_to_fill_margin)

    fill = kf.KCell(fill_name)
    fill.fill_region(
        region, fill_cell_index, fill_cell_box, None, region, fill_margin, None
    )
    gdspath_out = gdspath_out or gdspath
    fill.write(str(gdspath_out))

if __name__ == "__main__":
    c = gf.components.mzi(decorator=gf.add_padding)
    gdspath = c.write_gds("mzi_fill.gds")
    fill(
        gdspath,
        fill_layers=("WG",),
        layer_to_fill=gf.LAYER.PADDING,
    )
    gf.show(gdspath)

```

@sebastian-goeldi 
@SkandanC 
@tvt173

SkandanC, 2023-04-08:
Hello @joamatab,

This script works for me

```

"""Dummy fill to keep density constant using klayout."""
from __future__ import annotations

from typing import Optional, Tuple

import gdsfactory as gf
from gdsfactory.typings import PathType, LayerSpecs, Layer

import kfactory as kf
import klayout.db as kdb

def fill(
    gdspath,
    layer_to_fill: Layer,
    layer_to_fill_margin: float = 0,
    layers_to_avoid: LayerSpecs = None,
    layers_to_avoid_margin: float = 0,
    cell_name: Optional[str] = None,
    fill_cell_name: Optional[str] = None,
    fill_layers: LayerSpecs = None,
    fill_size: Tuple[float, float] = (10, 10),
    fill_spacing: Tuple[float, float] = (20, 20),
    fill_name: str = "fill",
    gdspath_out: Optional[PathType] = None,
) -> None:
    """Write gds file with fill.

    Args:
        gdspath: GDS input.
        gdspath_out: Optional GDS output. Defaults to input.
        cell_name: Optional cell to fill. Defaults to top cell.
        fill_cell_name: Optional cell name to use as fill.
        fill_layers:
        fill_size:
        fill_spacing:
        fill_name: name of the cell containing all fill cells.
    """

    lib = kf.kcell.KLib()
    lib.read(filename=str(gdspath))
    cell = lib[cell_name or 0]

    if fill_cell_name:
        fill_cell = lib[fill_cell_name]

    else:
        fill_cell = kf.KCell("fill_cell")
        for layer in fill_layers:
            layer = gf.get_layer(layer)
            layer = kf.klib.layer(*layer)
            fill_cell << kf.pcells.waveguide(
                width=fill_size[0], length=fill_size[1], layer=layer
            )

    fill_cell_index = fill_cell.cell_index()  # fill cell index
    fill_cell_box = fill_cell.bbox().enlarged(
        fill_spacing[0] / 2 * 1e3, fill_spacing[1] / 2 * 1e3
    )
    fill_margin = kf.kdb.Point(0, 0)

    layer_to_fill = gf.get_layer(layer_to_fill)
    layer_to_fill = cell.klib.layer(*layer_to_fill)
    region = kdb.Region()
    region_ = kdb.Region()
    for layer in cell.klib.layer_indices():
        region_.insert(cell.begin_shapes_rec(layer)) if layer != layer_to_fill else None
    
    region.insert(cell.begin_shapes_rec(layer_to_fill))
    region = region - region_

    fill = kf.KCell(fill_name)
    fill.fill_region(
        region, fill_cell_index, fill_cell_box, fill_margin,
    )
    gdspath_out = gdspath_out or gdspath
    for layer in cell.klib.layer_infos():
        fill.shapes(fill.klib.layer(layer)).insert(cell.begin_shapes_rec(cell.klib.layer(layer)))
    # fill.shapes(layer_to_fill).insert(cell.shapes(layer_to_fill))
    fill.write(str(gdspath_out))

if __name__ == "__main__":
    c = gf.components.mzi(decorator=gf.add_padding)
    gdspath = c.write_gds("mzi_fill.gds")
    fill(
        gdspath,
        fill_layers=("WG",),
        layer_to_fill=gf.LAYER.PADDING,
    )
    gf.show(gdspath)
```

sebastian-goeldi, 2023-04-08:
@SkandanC problem is that is single thread. Joaquin wants one with tiling processor

SkandanC, 2023-04-08:
I am aware, Joaquin's also mentioned that the script needs fixing.

joamatab, 2023-04-08:
yes, thank you Skandan and Sebastian

```python
"""Dummy fill to keep density constant using klayout."""
from __future__ import annotations

from typing import Optional, Tuple

import gdsfactory as gf
from gdsfactory.typings import PathType, LayerSpecs, Layer

import kfactory as kf
import klayout.db as kdb

def fill(
    gdspath,
    layer_to_fill: Layer,
    layer_to_fill_margin: float = 0,
    layers_to_avoid: LayerSpecs = None,
    layers_to_avoid_margin: float = 0,
    cell_name: Optional[str] = None,
    fill_cell_name: Optional[str] = None,
    fill_layers: LayerSpecs = None,
    fill_size: Tuple[float, float] = (10, 10),
    fill_spacing: Tuple[float, float] = (20, 20),
    fill_name: str = "fill",
    gdspath_out: Optional[PathType] = None,
) -> None:
    """Write gds file with fill.

    Args:
        gdspath: GDS input.
        gdspath_out: Optional GDS output. Defaults to input.
        cell_name: Optional cell to fill. Defaults to top cell.
        fill_cell_name: Optional cell name to use as fill.
        fill_layers:
        fill_size:
        fill_spacing:
        fill_name: name of the cell containing all fill cells.
    """

    lib = kf.kcell.KLib()
    lib.read(filename=str(gdspath))
    cell = lib[cell_name or 0]

    if fill_cell_name:
        fill_cell = lib[fill_cell_name]

    else:
        fill_cell = kf.KCell("fill_cell")
        for layer in fill_layers:
            layer = gf.get_layer(layer)
            layer = kf.klib.layer(*layer)
            fill_cell << kf.pcells.waveguide.waveguide(
                width=fill_size[0], length=fill_size[1], layer=layer
            )

    fill_cell_index = fill_cell.cell_index()  # fill cell index
    fill_cell_box = fill_cell.bbox().enlarged(
        fill_spacing[0] / 2 * 1e3, fill_spacing[1] / 2 * 1e3
    )
    fill_margin = kf.kdb.Point(0, 0)

    layer_to_fill = gf.get_layer(layer_to_fill)
    layer_to_fill = cell.klib.layer(*layer_to_fill)
    region = kdb.Region()
    region_ = kdb.Region()
    for layer in cell.klib.layer_indices():
        region_.insert(cell.begin_shapes_rec(layer)) if layer != layer_to_fill else None

    region.insert(cell.begin_shapes_rec(layer_to_fill))
    region = region - region_

    fill = kf.KCell(fill_name)
    fill.fill_region(
        region,
        fill_cell_index,
        fill_cell_box,
        fill_margin,
    )
    gdspath_out = gdspath_out or gdspath
    for layer in cell.klib.layer_infos():
        fill.shapes(fill.klib.layer(layer)).insert(
            cell.begin_shapes_rec(cell.klib.layer(layer))
        )
    # fill.shapes(layer_to_fill).insert(cell.shapes(layer_to_fill))
    fill.write(str(gdspath_out))

@gf.cell
def cell_with_pad():
    c = gf.Component()
    c << gf.components.mzi(decorator=gf.add_padding)
    pad = c << gf.components.pad(size=(2, 2))
    pad.movey(10)
    return c

if __name__ == "__main__":
    c = cell_with_pad()
    c.show()
    gdspath = c.write_gds("mzi_fill.gds")
    fill(
        gdspath,
        fill_layers=("WG",),
        layer_to_fill=gf.LAYER.PADDING,
        fill_cell_name="pad_size2__2",
        fill_spacing=(1, 1),
        fill_size=(1, 1),
    )
    gf.show(gdspath)

```

how could we use a specific cell for the fill?

sebastian-goeldi, 2023-04-08:
@joamatab  Just wrote the fill with tilingprocessor:

https://github.com/gdsfactory/kfactory/blob/drc/src/kfactory/utils/geo/fill.py

a minimal example how to use it:

```python
c = kf.KCell("ToFill")
c.shapes(kf.klib.layer(1, 0)).insert(
    kf.kdb.DPolygon.ellipse(kf.kdb.DBox(5000, 3000), 512)
)
c.shapes(kf.klib.layer(10, 0)).insert(
    kf.kdb.DPolygon(
        [kf.kdb.DPoint(0, 0), kf.kdb.DPoint(5000, 0), kf.kdb.DPoint(5000, 3000)]
    )
)
c

fc = kf.KCell("fill")
fc.shapes(fc.klib.layer(2, 0)).insert(kf.kdb.DBox(20, 40))
fc.shapes(fc.klib.layer(3, 0)).insert(kf.kdb.DBox(30, 15))
fc

import kfactory.utils.geo.fill as fill

# fill.fill_tiled(c, fc, [(kf.klib.layer(1,0), 0)], exclude_layers = [(kf.klib.layer(10,0), 100), (kf.klib.layer(2,0), 0), (kf.klib.layer(3,0),0)], x_space=5, y_space=5)
fill.fill_tiled(
    c,
    fc,
    [(kf.klib.layer(1, 0), 0)],
    exclude_layers=[
        (kf.klib.layer(10, 0), 100),
        (kf.klib.layer(2, 0), 0),
        (kf.klib.layer(3, 0), 0),
    ],
    x_space=5,
    y_space=5,
)

c
```

It's also in the WIP DRC notebook https://github.com/gdsfactory/kfactory/blob/drc/docs/notebooks/02_DRC.py

joamatab, 2023-04-09:
Looks great Sebastian,

why did the code above  produced `┃[Process exited 139]` Segfault, when i try to specify a layer

joamatab, 2023-04-09:
why did my code above produced ┃[Process exited 139] Segfault, when i try to specify a layer

```python

"""Dummy fill to keep density constant using klayout."""
from __future__ import annotations

from typing import Optional, Tuple

import gdsfactory as gf
from gdsfactory.typings import PathType, LayerSpecs, LayerSpec

import kfactory as kf
import klayout.db as kdb

def fill(
    gdspath,
    layer_to_fill: LayerSpec,
    layer_to_fill_margin: float = 0,
    layers_to_avoid: Optional[Tuple[LayerSpec, float], ...] = None,
    cell_name: Optional[str] = None,
    fill_cell_name: str = "fill_cell",
    create_new_fill_cell: bool = False,
    keep_shapes: bool = False,
    fill_layers: LayerSpecs = None,
    fill_size: Tuple[float, float] = (10, 10),
    fill_spacing: Tuple[float, float] = (20, 20),
    fill_name: str = "fill",
    gdspath_out: Optional[PathType] = None,
) -> None:
    """Write gds file with fill.

    Args:
        gdspath: GDS input.
        layer_to_fill: Layer that defines the region to fill.
        layer_to_fill_margin: in um.
        layers_to_avoid: Layer to avoid to margin ((LAYER.WG, 3.5), (LAYER.SLAB, 2)).
        cell_name: Optional cell to fill. Defaults to top cell.
        fill_cell_name: Optional cell name to use as fill.
        create_new_fill_cell: creates new fill cell, otherwise uses fill_cell_name from gdspath.
        keep_shapes: keep gds shapes.
        fill_layers: if create_new_fill_cell=True, defines new fill layers.
        fill_size: if create_new_fill_cell=True, defines new fill size.
        fill_spacing: fill pitch in x and y.
        fill_name: name of the cell containing all fill cells.
        gdspath_out: Optional GDS output. Defaults to input.
    """

    lib = kf.kcell.KLib()
    lib.read(filename=str(gdspath))
    cell = lib[cell_name or 0]

    fill = kf.KCell(fill_name)

    if create_new_fill_cell:
        if lib.has_cell(fill_cell_name):
            raise ValueError(f"{fill_cell_name!r} already in {str(gdspath)!r}")
        fill_cell = kf.KCell(fill_cell_name)
        for layer in fill_layers:
            layer = gf.get_layer(layer)
            layer = kf.klib.layer(*layer)
            fill_cell << kf.pcells.waveguide.waveguide(
                width=fill_size[0], length=fill_size[1], layer=layer
            )
    else:
        fill_cell = lib[fill_cell_name]

    fill_cell_index = fill_cell.cell_index()  # fill cell index
    fill_cell_box = fill_cell.bbox().enlarged(
        fill_spacing[0] / 2 * 1e3, fill_spacing[1] / 2 * 1e3
    )
    fill_margin = kf.kdb.Point(0, 0)

    layer_to_fill = gf.get_layer(layer_to_fill)
    layer_to_fill = cell.klib.layer(*layer_to_fill)
    region = kdb.Region()
    region_avoid_all = kdb.Region()

    if layers_to_avoid:
        for layer, margin in layers_to_avoid:
            layer = gf.get_layer(layer)
            layer = kf.klib.layer(*layer)
            region_avoid = kdb.Region()
            region_avoid.insert(
                cell.begin_shapes_rec(layer)
            ) if layer != layer_to_fill else None
            region_avoid = region_avoid.size(margin * 1e3)
            region_avoid_all = region_avoid_all + region_avoid

    region.insert(cell.begin_shapes_rec(layer_to_fill))
    region.size(-layer_to_fill_margin * 1e3)
    region_to_fill = region - region_avoid_all

    fill.fill_region(
        region_to_fill,
        fill_cell_index,
        fill_cell_box,
        fill_margin,
    )

    if keep_shapes:
        for layer in cell.klib.layer_infos():
            fill.shapes(fill.klib.layer(layer)).insert(
                cell.begin_shapes_rec(cell.klib.layer(layer))
            )

    gdspath_out = gdspath_out or gdspath
    fill.write(str(gdspath_out))

@gf.cell
def cell_with_pad():
    c = gf.Component()
    c << gf.components.mzi(decorator=gf.add_padding)
    pad = c << gf.components.pad(size=(2, 2))
    pad.movey(10)
    return c

if __name__ == "__main__":
    c = cell_with_pad()
    c.show()
    gdspath = c.write_gds("mzi_fill.gds")
    # fill(
    #     gdspath,
    #     fill_layers=("WG",),
    #     layer_to_fill=gf.LAYER.PADDING,
    #     layers_to_avoid=((gf.LAYER.WG, 0),),
    #     fill_cell_name="fill_cell",
    #     create_new_fill_cell=True,
    #     fill_spacing=(1, 1),
    #     fill_size=(1, 1),
    #     layer_to_fill_margin=25,
    # )
    fill(
        gdspath,
        fill_layers=("WG",),
        layer_to_fill=gf.LAYER.PADDING,
        fill_cell_name="pad_size2__2",
        create_new_fill_cell=False,
        fill_spacing=(1, 1),
        fill_size=(1, 1),
    )
    gf.show(gdspath)

```

sebastian-goeldi, 2023-04-09:
Because you are doing the same thing as always ;) . You are creating a `lib = kf.KLib()` and then you are dumping into the standard lib of `kf.klib` on a layer that doesn't exist

joamatab, 2023-04-09:
i see

this still fails

```python
"""Dummy fill to keep density constant using klayout."""
from __future__ import annotations

from typing import Optional, Tuple

import gdsfactory as gf
from gdsfactory.typings import PathType, LayerSpecs, LayerSpec

import kfactory as kf
import klayout.db as kdb

def fill(
    gdspath,
    layer_to_fill: LayerSpec,
    layer_to_fill_margin: float = 0,
    layers_to_avoid: Optional[Tuple[LayerSpec, float], ...] = None,
    cell_name: Optional[str] = None,
    fill_cell_name: str = "fill_cell",
    create_new_fill_cell: bool = False,
    keep_shapes: bool = False,
    fill_layers: LayerSpecs = None,
    fill_size: Tuple[float, float] = (10, 10),
    fill_spacing: Tuple[float, float] = (20, 20),
    fill_name: str = "fill",
    gdspath_out: Optional[PathType] = None,
) -> None:
    """Write gds file with fill.

    Args:
        gdspath: GDS input.
        layer_to_fill: Layer that defines the region to fill.
        layer_to_fill_margin: in um.
        layers_to_avoid: Layer to avoid to margin ((LAYER.WG, 3.5), (LAYER.SLAB, 2)).
        cell_name: Optional cell to fill. Defaults to top cell.
        fill_cell_name: Optional cell name to use as fill.
        create_new_fill_cell: creates new fill cell, otherwise uses fill_cell_name from gdspath.
        keep_shapes: keep gds shapes.
        fill_layers: if create_new_fill_cell=True, defines new fill layers.
        fill_size: if create_new_fill_cell=True, defines new fill size.
        fill_spacing: fill pitch in x and y.
        fill_name: name of the cell containing all fill cells.
        gdspath_out: Optional GDS output. Defaults to input.
    """

    lib = kf.KLib()
    lib.read(filename=str(gdspath))
    cell = lib[cell_name or 0]

    c = kf.KCell(fill_name)

    if create_new_fill_cell:
        if lib.has_cell(fill_cell_name):
            raise ValueError(f"{fill_cell_name!r} already in {str(gdspath)!r}")
        fill_cell = kf.KCell(fill_cell_name)
        for layer in fill_layers:
            layer = gf.get_layer(layer)
            layer = kf.klib.layer(*layer)
            fill_cell << kf.pcells.waveguide.waveguide(
                width=fill_size[0], length=fill_size[1], layer=layer
            )
    else:
        fill_cell = lib[fill_cell_name]

    fill_cell_index = fill_cell.cell_index()  # fill cell index
    fill_cell_box = fill_cell.bbox().enlarged(
        fill_spacing[0] / 2 * 1e3, fill_spacing[1] / 2 * 1e3
    )
    fill_margin = kf.kdb.Point(0, 0)

    layer_to_fill = gf.get_layer(layer_to_fill)
    layer_to_fill = cell.klib.layer(*layer_to_fill)
    region = kdb.Region()
    region_avoid_all = kdb.Region()

    if layers_to_avoid:
        for layer, margin in layers_to_avoid:
            layer = gf.get_layer(layer)
            layer = kf.klib.layer(*layer)
            region_avoid = kdb.Region()
            region_avoid.insert(
                cell.begin_shapes_rec(layer)
            ) if layer != layer_to_fill else None
            region_avoid = region_avoid.size(margin * 1e3)
            region_avoid_all = region_avoid_all + region_avoid

    region.insert(cell.begin_shapes_rec(layer_to_fill))
    region.size(-layer_to_fill_margin * 1e3)
    region_to_fill = region - region_avoid_all

    c.fill_region(
        region_to_fill,
        fill_cell_index,
        fill_cell_box,
        fill_margin,
    )

    if keep_shapes:
        for layer in cell.klib.layer_infos():
            c.shapes(c.klib.layer(layer)).insert(
                cell.begin_shapes_rec(cell.klib.layer(layer))
            )

    gdspath_out = gdspath_out or gdspath
    c.write(str(gdspath_out))

@gf.cell
def cell_with_pad():
    c = gf.Component()
    c << gf.components.mzi(decorator=gf.add_padding)
    pad = c << gf.components.pad(size=(2, 2))
    pad.movey(10)
    return c

if __name__ == "__main__":
    c = cell_with_pad()
    c.show()
    gdspath = c.write_gds("mzi_fill.gds")

    use_fill_cell = False  # Works
    use_fill_cell = True  # Segfaults

    if use_fill_cell:
        fill(
            gdspath,
            fill_layers=("WG",),
            layer_to_fill=gf.LAYER.PADDING,
            fill_cell_name="pad_size2__2",
            create_new_fill_cell=False,
            fill_spacing=(1, 1),
            fill_size=(1, 1),
        )
    else:
        fill(
            gdspath,
            fill_layers=("WG",),
            layer_to_fill=gf.LAYER.PADDING,
            layers_to_avoid=((gf.LAYER.WG, 0),),
            fill_cell_name="fill_cell",
            create_new_fill_cell=True,
            fill_spacing=(1, 1),
            fill_size=(1, 1),
            layer_to_fill_margin=25,
        )
    gf.show(gdspath)

```

joamatab, 2023-04-09:
nice, thank you! it works great!

sebastian-goeldi, 2023-04-10:
> Very good :) 
> 
> In version 0.28.6 a bug was fixed that may be the reason for the fill discontinuity observed above: https://github.com/KLayout/klayout/issues/1309.

Thank you @klayoutmatthias ! The issue here is (I am pretty sure) a rounding of the float. The gap is exactly 1nm too small for a fill cell (rounded .999 instead of 1 or similar). I can prepare a pure klayout testcase of this if it helps. But I will try with a klayout assembled from master first and let you know whether it works.

Also, is that space check syntax available to the tilingprocessor expressions?
That would be awesome ;) I added one for fixing of boolean layers etc that can have nasty errors on case of bends meeting a straight or another bend

sebastian-goeldi, 2023-04-10:
I will happily try that!

Yes, I am fully aware of that. That's why I didn't open an issue or investigate the details, yet. In photonics we have seen discontinuities like these affecting the devices' performance if they land at unfortunate places (in those cases they were man made usually though). Therefore, it would be nice to avoid, but we have solutions to avoid it should we ever run into a problem like this from rounding ^^. Though it might never happen because normally there isn't this much continuous space free.

closed 2023-04-26 by joamatab