gdsfactory/gdsfactory #1536: Create a slim installation for GDS factory that supports only the basic layout functionality

atorkmabrains, 2023-04-10:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
GDS Factory requires a lot of packages that we might avoid needing to make is portable and easy to use with other applications. We need to create an installable version may be gdsfactory-slim that only has the layout functionality and doesn't require any other packages related to simulation. This would be really helpful for using gdsfactory in the backend.

The requested features required in this slim package is anything documented in the layout section.

tvt173, 2023-04-10:
i think this is already the case when you `pip install gdsfactory` no? to get the other dependencies for simulation etc. you have to explicitly specify, i.e. `pip install gdsfactory[devsim]`

Docs
https://gdsfactory.github.io/gdsfactory/index.html#install-gdsfactory-plugins
Source
https://github.com/gdsfactory/gdsfactory/blob/main/pyproject.toml#L23

atorkmabrains, 2023-04-10:
@tvt173 I think it comes with more than was is needed. We need a slimmer version.

joamatab, 2023-08-15:
we moved all the complicated installation plugins into

https://github.com/gdsfactory/gplugins

closed 2023-08-15 by joamatab